alf45tar/PedalinoMini #362: Installing Pedalino on TTGO TT-Display

SrMorrris, 2022-08-25:
Connect to pc and installing via web..everything works fine..
But in vscode-platformio i can´t upload..

Everything its fine at building (file system image too).

But i can´t upload (file system image too).

I can go to web-config and everything its fine, i have assumed this problem doesn´t matter, but i want to ask you about that..i don´t know how to solve.
Ok,thanks..

alf45tar, 2022-08-25:
Without any other information (like errors messages) I can only suppose you didn't select the correct serial port for upload.

SrMorrris, 2022-08-25:
Ok,Excuse me...are my first days on vscode ...
So i have selected the TTGO-TDisplay environment in platformio
And then I have changed this two options on PROJECTS/CONFIG/TTGO-TDisplay 
[url=https://ibb.co/whvdsnR][img]https://i.ibb.co/xq42HnD/2.png[/img][/url]
Now it´s solved, thanks

SrMorrris, 2022-08-25:
> [image]

closed 2022-08-26 by SrMorrris