alf45tar/PedalinoMini #364: battery voltage fluctuation

bobvc133, 2022-09-01:
I use a 2800mAh 18650 battery. During use, I often find that the voltage fluctuates greatly. Sometimes the voltage display is low, and the screen flickers. After restarting, sometimes it returns to normal voltage, and sometimes it cannot be charged. I think it may be the TTGO motherboard or There is a problem with power supply management, I don't know if you have encountered it?
[image]

alf45tar, 2022-09-01:
I didn't encounter it. The discharge is usually very linear. Did you try a different battery?

alf45tar, 2022-09-01:
I have the same issue with WiFi and Bluetooth activated.
There is no fluctuation with only Wifi or only Bluetooth.
There is no fluctuation also if you power off and on. But if you reset the board the fluctuation return.
It needs more investigation.

alf45tar, 2022-09-01:
Check the battery assembly. I solved it reinstalling the battery in the slot. In my case was a loose assembly and poor connection.

bobvc133, 2022-09-01:
The problem of poor contact is basically eliminated, and I have replaced the cable and battery many times. At first I thought it was because the battery voltage was too low, but later I found out that the same problem occurs after fully charged. Sometimes during the booting process, it cannot be turned on due to frequent restarts, and the display flashes brightly and darkly, as if the power is insufficient. After re-plugging and unplugging the battery, there is a certain probability that it will suddenly return to normal, and the power display is sufficient and there is basically no fluctuation. But the problem persists after turning the power off and on again

bobvc133, 2022-09-03:
> Check the battery assembly. I solved it reinstalling the battery in the slot. In my case was a loose assembly and poor connection.

I replaced all the connected components, and checked the resistance with a multimeter. It turned out that the voltage fluctuation problem was caused by poor internal contact of the power switch, and the quality of the power switch was not good enough.

closed 2022-09-04 by bobvc133