alf45tar/PedalinoMini #366: External button

bobvc133, 2022-09-12:
I want to connect two external buttons on ESP32_DOIT, one to control the configuration file switch, one to control the library switch, GPIO0 has been mapped to GPIO23 and can be used normally, and I also want to set an 8th button similar to GPIO35 of TTGO T DISPALY, which can be used GPIO2 interface, what code needs to be modified?
[image]

bobvc133, 2022-09-24:
I tested and found that when GPIO23 is mapped as the BOOT button, and the power on/off function is set, it can be powered off but cannot be powered on

alf45tar, 2022-09-28:
Not completely clear your request. Let me try to reply.

Configuration file switch is not possible using pedals. This is not an available action. It is not so complex to implement. Do you need it?

What do you mean with library switch? Is it he same of bank switch? if yest it is already implemented as an action and can be linked to any pedal.

What do yo mean with GPIO23 mapped as BOOT button? BOOT button is connected to GPIO00.

bobvc133, 2022-09-28:
What I mean is to ask which extra GPIOs can be used on the ESP32_DOIT development board. I plan to connect two external buttons to implement the functions I defined for GPIO0 and GPIO35 on the TTGO development board.
[image]

bobvc133, 2022-12-03:
This problem has been solved, modify the code to change GPIO0 to GPIO2, add a pull-down resistor, and then add GPIO23 port as pedal 8
[image]

closed 2022-12-03 by bobvc133