alf45tar/PedalinoMini #368: When I choose options tab via web interface device rebooting

e36910, 2022-09-15:
Device rebooting after selected options, or if I write link 192.168.4.1/options. What can I do?

rigr, 2022-09-15:
Had the same issue while the unit was powered by USB. A separate ower source made it stable.

e36910, 2022-09-15:
I'm using power via VIN and GDN pins wiht 5v phone adapters

rigr, 2022-09-15:
The 5V phone adapter might be too weak if you power the LEDs from the same source. But maybe there is another issue.what board do you use? what version of pedalino? How many LEDs? what interfaces are enabled? etc

alf45tar, 2022-09-17:
May I suggest to reset to factory default or Upload Filesystem Image?

e36910, 2022-09-18:
After update the device does not turn on

alf45tar, 2022-10-15:
I think I cannot help without more details. Reopen it if you need anything else.

closed 2022-10-15 by alf45tar