alf45tar/PedalinoMini #369: RGB order is lost after reboot and config is not saving

FelixMuellCode, 2022-09-23:
The RGB order is lost after rebooting everything else is properly saved.

When I try to save a config, a file with wrong syntax is created.
There is no error on the serial output.

I'm using a lolin32 lite board which I flash using the ESP32Devkit1 Environment, this could be a problem.

<details><summary>cfg</summary>

```
{"Globals":[{"Hostname":"Pedalino","BootMode":1,"Profile":0,"SSID":"Beacon","WiFiPassword":"Maxdome3.0","SSIDsoftAP":"Pedalino","PasswordSoftAP":"","HTTPUsername":"","HTTPPassword":"","Theme":"bootstrap","PressTime":200,"DoublePressTime":400,"LongPressTime":500,"RepeatPressTime":500,"EncoderSesitivity":10,"FlipScreen":true,"TapDanceMode":false,"RepeatOnBankSwitch":false,"TapDanceBank":true,"Leds":10,"RGBOrder":"RGB","LedsOnBrightness":255,"LedsOffBrightness":255}],"Ladder":[{"Step":1,"Level":0},{"Step":2,"Level":113},{"Step":3,"Level":310},{"Step":4,"Level":498},{"Step":5,"Level":733},{"Step":6,"Level":897},{"Step":7,"Level":1023}],"Pedals":[{"Pedal":1,"Mode":"Ladder","InvertPolarity":false,"PressMode":"1","AnalogResponse":"Linear","Min":0,"Max":0,"Easing":0.001,"ActivityThreshold":2,"AutoSensing":false,"LedButton0":0,"LedButton1":0,"LedButton2":0,"LedButton3":0,"LedButton4":0,"LedButton5":0},{"Pedal":2,"Mode":"Analog+Momentary","InvertPolarity":true,"PressMode":"1","AnalogResponse":"Linear","Min":1023,"Max":972,"Easing":0.100000001,"ActivityThreshold":256,"AutoSensing":true,"LedButton0":0,"LedButton1":0,"LedButton2":0,"LedButton3":0,"LedButton4":0,"LedButton5":0},{"Pedal":3,"Mode":"Analog+Momentary","InvertPolarity":true,"PressMode":"1","AnalogResponse":"Linear","Min":1023,"Max":972,"Easing":0.100000001,"ActivityThreshold":256,"AutoSensing":true,"LedButton0":0,"LedButton1":0,"LedButton2":0,"LedButton3":0,"LedButton4":0,"LedButton5":0},{"Pedal":4,"Mode":"Jog Wheel","InvertPolarity":false,"PressMode":"1","AnalogResponse":"Linear","Min":0,"Max":0,"Easing":0.001,"ActivityThreshold":2,"AutoSensing":false,"LedButton0":0,"LedButton1":0,"LedButton2":0,"LedButton3":0,"LedButton4":0,"LedButton5":0},{"Pedal":5,"Mode":"Momentary 1","InvertPolarity":false,"PressMode":"1","AnalogResponse":"Linear","Min":0,"Max":0,"Easing":0.001,"ActivityThreshold":2,"AutoSensing":false,"LedButton0":0,"LedButton1":0,"LedButton2":0,"LedButton3":0,"LedButton4":0,"LedButton5":0},{"Pedal":6,"Mode":"Momentary 1","InvertPolarity":false,"PressMode":"1","AnalogResponse":"Linear","Min":0,"Max":0,"Easing":0.001,"ActivityThreshold":2,"AutoSensing":false,"LedButton0":0,"LedButton1":0,"LedButton2":0,"LedButton3":0,"LedButton4":0,"LedButton5":0},{"Pedal":7,"Mode":"Momentary 1","InvertPolarity":false,"PressMode":"1","AnalogResponse":"Linear","Min":0,"Max":0,"Easing":0.001,"ActivityThreshold":2,"AutoSensing":false,"LedButton0":0,"LedButton1":0,"LedButton2":0,"LedButton3":0,"LedButton4":0,"LedButton5":0}],"BankNames":[{"Bank":0,"Name":"Global"},{"Bank":1,"Name":""},{"Bank":2,"Name":""},{"Bank":3,"Name":""},{"Bank":4,"Name":""},{"Bank":5,"Name":""},{"Bank":6,"Name":""},{"Bank":7,"Name":""},{"Bank":8,"Name":""},{"Bank":9,"Name":""},{"Bank":10,"Name":""},{"Bank":11,"Name":""},{"Bank":12,"Name":""},{"Bank":13,"Name":""},{"Bank":14,"Name":""},{"Bank":15,"Name":""},{"Bank":16,"Name":""},{"Bank":17,"Name":""},{"Bank":18,"Name":""},{"Bank":19,"Name":""},{"Bank":20,"Name":""}],"Actions":[{"Bank":0,"Pedal":7,"Button":1,"Led":0,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":8,"Event":"Click","Message":"Profile+","Channel":1,"Code":0,"Value1":1,"Value2":3},{"Bank":0,"Pedal":7,"Button":1,"Led":0,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":8,"Event":"Double Click","Message":"Profile-","Channel":1,"Code":0,"Value1":0,"Value2":127},{"Bank":0,"Pedal":7,"Button":1,"Led":0,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":8,"Event":"Long Press","Message":"Device Info","Channel":1,"Code":0,"Value1":0,"Value2":127},{"Bank":1,"Pedal":1,"Button":1,"Led":1,"Color0":"#ff00ff","Color1":"#ff00ff","NameOff":"","NameOn":"","Slot":0,"Event":"Press","Message":"Note On","Channel":1,"Code":1,"Value1":125,"Value2":0},{"Bank":1,"Pedal":1,"Button":1,"Led":1,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":0,"Event":"Release","Message":"Note Off","Channel":1,"Code":1,"Value1":0,"Value2":0},{"Bank":1,"Pedal":1,"Button":2,"Led":6,"Color0":"#fbff00","Color1":"#fbff00","NameOff":"","NameOn":"","Slot":0,"Event":"Press","Message":"Note On","Channel":1,"Code":2,"Value1":125,"Value2":0},{"Bank":1,"Pedal":1,"Button":2,"Led":6,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":0,"Event":"Release","Message":"Note Off","Channel":1,"Code":2,"Value1":0,"Value2":0},{"Bank":1,"Pedal":1,"Button":3,"Led":7,"Color0":"#e1ff00","Color1":"#ffff00","NameOff":"","NameOn":"","Slot":0,"Event":"Press","Message":"Note On","Channel":1,"Code":3,"Value1":125,"Value2":0},{"Bank":1,"Pedal":1,"Button":3,"Led":7,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":0,"Event":"Release","Message":"Note Off","Channel":1,"Code":3,"Value1":0,"Value2":0},{"Bank":1,"Pedal":1,"Button":4,"Led":5,"Color0":"#00ff00","Color1":"#02ff00","NameOff":"","NameOn":"","Slot":0,"Event":"Press","Message":"Note On","Channel":1,"Code":4,"Value1":125,"Value2":0},{"Bank":1,"Pedal":1,"Button":4,"Led":5,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":0,"Event":"Release","Message":"Note Off","Channel":1,"Code":4,"Value1":0,"Value2":0},{"Bank":1,"Pedal":1,"Button":5,"Led":4,"Color0":"#0000ff","Color1":"#0000ff","NameOff":"","NameOn":"","Slot":0,"Event":"Press","Message":"Note On","Channel":1,"Code":5,"Value1":125,"Value2":0},{"Bank":1,"Pedal":1,"Button":5,"Led":4,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":8,"Event":"Release","Message":"Note Off","Channel":1,"Code":5,"Value1":0,"Value2":0},{"Bank":1,"Pedal":1,"Button":6,"Led":3,"Color0":"#ff0000","Color1":"#ff0000","NameOff":"","NameOn":"","Slot":8,"Event":"Press","Message":"Note On","Channel":1,"Code":6,"Value1":125,"Value2":0},{"Bank":1,"Pedal":1,"Button":6,"Led":255,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":8,"Event":"Press","Message":"Note On","Channel":1,"Code":6,"Value1":125,"Value2":0},{"Bank":1,"Pedal":1,"Button":6,"Led":3,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":8,"Event":"Release","Message":"Note Off","Channel":1,"Code":6,"Value1":0,"Value2":0},{"Bank":1,"Pedal":2,"Button":1,"Led":2,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":8,"Event":"Press","Message":"Note On","Channel":1,"Code":7,"Value1":0,"Value2":0},{"Bank":1,"Pedal":2,"Button":1,"Led":2,"Color0":"#ff0000","Color1":"#ff0000","NameOff":"","NameOn":"","Slot":8,"Event":"Release","Message":"Note Off","Channel":1,"Code":7,"Value1":100,"Value2":127},{"Bank":1,"Pedal":2,"Button":1,"Led":255,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":0,"Event":"Move","Message":"Note On","Channel":1,"Code":36,"Value1":0,"Value2":127},{"Bank":1,"Pedal":3,"Button":1,"Led":0,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":0,"Event":"Press","Message":"Note On","Channel":1,"Code":1,"Value1":0,"Value2":0},{"Bank":1,"Pedal":3,"Button":1,"Led":255,"Color0":"#000000","Color1":"#000000","NameOff":"","NameOn":"","Slot":8,"Event":"Release","Message":"Note Off","Channel":1,"Code":0,"Value1":100,"Value2":127},{"Bank":1,"Pedal":3,"Button":1,"Led":0,"Color0":"#000000","Color1":"#000000","NameOff":"","Name

```

</details>

Thanks.

[image]
[image]

alf45tar, 2022-09-23:
RGB order not saved bug is already fixed in y internal version to be released shortly.
Do you have enough space on SPIFFS to save the configuration? You can check oh PedalinoMini home page.

alf45tar, 2022-09-23:
Just released public version 2.5.2 with RGB order bug fixed.

FelixMuellCode, 2022-09-24:
Awsome thanks.

It should be enough

```
PSRAM Used/Total
0/0 kB
NVS Used/Total
671/2016 entries
SPIFFS Used/Total
178/228 kB
Free Heap Size (Max Allocation)
99 kB (64 kB)
```

2.5.2 is not working at all on the lolin32lite.
It's forced resetting when trying to load a config. I think the filesystem is too different.

Lolin32lite support would be awesome as these boards can charge a battery while only costing 2,8€ on AliExpress.
I got it working by only changing in the OLED Config.
```
#define OLED_I2C_SDA      23//SDA
#define OLED_I2C_SCL      19//SCL

```

FelixMuellCode, 2022-09-24:
With the new version everything config and LED related works perfectly.

Lolin32lite support would be awesome as these boards can charge a battery while only costing 2,8€ on AliExpress.
I got it working by only changing in the OLED Config. Unfortunately only every second horizontal display line is shown, the rest is lost.

```
#define OLED_I2C_SDA      23//SDA
#define OLED_I2C_SCL      19//SCL
```

MiqViq, 2022-10-08:
I have the same issue concerning RGB order lost after reboot.
I cannot find the discussed version  2.5.2 in the releases.

alf45tar, 2022-10-08:
Version 2.5.2 is available into the public repository even if I am not providing a precompiled firmware yet.

alf45tar, 2022-10-15:
Precompiled firmware for version 2.5.2 released

closed 2022-10-15 by alf45tar