alf45tar/PedalinoMini #370: Can pedalino connecting with chocolete midi bluetooth adapter?

e36910, 2022-09-24:
I found a video showing how to connect another device to guitar processors via bluetooth - [https://youtu.be/7gNBT2MgySM
[image]
](https://youtu.be/7gNBT2MgySM)
I have 2 qestions:
1. Is it possible to connect an arduino device to this midi bluetooth device as shown in the video, if yes, how?
2. Is it possible to make an arduino bluetooth receiver that will connect to guitar processors via usb?

alf45tar, 2022-10-17:
1. Yes, it is. For example using this library https://github.com/lathoub/Arduino-BLE-MIDI
2. Yes, it is. If the guitar processor is a usb host is a little bit more easy.

alf45tar, 2022-10-17:
In the next version under testing right now, PedalinoMini will connect to a second PedalinoMini via bluetooth. one PedalinoMini will act as server and the second one as a client.
You could use a bluetooth connection to transport midi messages from one legacy midi device to another legacy midi device.

alf45tar, 2022-11-06:
Just released version 3.0.0 which includes BLE client.

closed 2022-11-06 by alf45tar