alf45tar/PedalinoMini #372: ENHANCEMENT: Detecting simultaneous press/hold of 2 buttons?

MiqViq, 2022-10-09:
Does PedalinoMini detect simultaneous press/hold of 2 buttons?
I think it could be a very useful feature to have.

And thanks for your hard work on this project!

-MiqViq

alf45tar, 2022-10-09:
No, it doesn't.

alf45tar, 2022-11-06:
Just released version 3.0.0 which includes simultaneous press.

closed 2022-11-06 by alf45tar

bobvc133, 2022-11-07:
> Just released version 3.0.0 which includes simultaneous press.

Pressing at the same time is like pressing, long pressing, and pressing repeatedly. Because of the difference in the response time of pressing and the reason for the asynchrony, it is easy to perform the action of pressing first, and the probability of misoperation is high.

alf45tar, 2022-11-09:
You are right. Simultaneous pressing is not exactly simultaneous because it is quite impossible by a human to be so precise but the implementation take care of it. 
Two buttons are considered simultaneous pressed if the second is pressed within 50 ms (PED_SIMULTANEOUS_GAP) from the first. The actions linked to single press of first button are delayed up to 50 ms waiting the second button press. Singles press actions of both buttons never happen if they are pressed with maximum 50 ms of difference. Of course if you are not fast enough single press actions are triggered. Increasing PED_SIMULTANEOUS_GAP give user more time but it delays the actions on single press.
Actions delay do not happen if the button is not involved in any simultaneous control.