alf45tar/PedalinoMini #376: Chip resistors

bobvc133, 2022-10-12:
Can the 1/4W resistor in the schematic be replaced with a chip resistor in a 0603 package? This saves board space

alf45tar, 2022-10-12:
No experience with 0603 resistor but it looks like they are 1/10W.
They are ok for 10K pull-up resistors but not for 220 ohm used in MIDI because they consume 0,11W.

smadds, 2022-10-12:
1/10W should be fine, as this is not a continuous 5V DC. Actual power is likely to be about half of the max

alf45tar, 2022-10-12:
I agree it can works but to be safe it is better to use 2 resistors because when the UART signal is idle, it sits at a logic high state.
The value of the 2 resistors should be 1/2 if connected in serie or 2x if connected in parallel.

bobvc133, 2022-10-13:
There should be no problem using a 1/4W 1206 resistor, the volume is slightly larger

alf45tar, 2022-10-14:
1/4W resistor are fine.

closed 2022-10-14 by alf45tar