alf45tar/PedalinoMini #377: Anything changed with Ladder config on later versions?

Designbase, 2022-10-15:
Hi,

I recently updated the firmware from 2.3.2 to 2.5.1 and now my ladder configured buttons do not work anymore.
Redid buttons via breadboard to exclude any mistakes and I can calibrate the ladder, all 6 buttons work on calibration but if I add actions to the buttons only the first one will work. Also tested with different ESP32 Boards.

Am I missing something that's changed over time? It's been some time since I worked on this project.

Thanks, Ron

alf45tar, 2022-10-15:
Can you check the log if the button pressure and release is logged?

Designbase, 2022-10-15:
Mmmmm... it says everywhere Pedal: 1... is that the isse?

**This is what I am getting in the Serial Monitor:**

Pedal: 1     Button: 1    EventType: 0     ButtonState: 0
Action: 
CONTROL CHANGE.....Code 100......Value   0.....Channel  1
Pedal: 1     Button: 1    EventType: 1     ButtonState: 1
Pedal: 1     Button: 2    EventType: 0     ButtonState: 0
Pedal: 1     Button: 2    EventType: 1     ButtonState: 1
Pedal: 1     Button: 2    EventType: 2     ButtonState: 1
Pedal: 1     Button: 3    EventType: 0     ButtonState: 0
Pedal: 1     Button: 3    EventType: 1     ButtonState: 1
Pedal: 1     Button: 4    EventType: 0     ButtonState: 0
Pedal: 1     Button: 4    EventType: 1     ButtonState: 1
Pedal: 1     Button: 4    EventType: 2     ButtonState: 1
Pedal: 1     Button: 5    EventType: 0     ButtonState: 0
Pedal: 1     Button: 5    EventType: 1     ButtonState: 1
Pedal: 1     Button: 5    EventType: 2     ButtonState: 1
Pedal: 1     Button: 6    EventType: 0     ButtonState: 0
Pedal: 1     Button: 6    EventType: 1     ButtonState: 1
Pedal: 1     Button: 6    EventType: 2     ButtonState: 1
Pedal: 7     Button: 1    EventType: 0     ButtonState: 0
Action: 
Pedal: 7     Button: 1    EventType: 1     ButtonState: 1
Pedal: 7     Button: 1    EventType: 0     ButtonState: 0
Action: 
Pedal: 7     Button: 1    EventType: 1     ButtonState: 1
Pedal: 7     Button: 1    EventType: 2     ButtonState: 1
Pedal: 8     Button: 1    EventType: 0     ButtonState: 0
Pedal: 8     Button: 1    EventType: 1     ButtonState: 1
Pedal: 8     Button: 1    EventType: 2     ButtonState: 1
Action: 
Updating NVS Profile A ...
[NVS][Current Bank]:  1
done
Loading profile 1 ...
Reading NVS Profile B ... done
Bank  2
Pedal  1        1   MOMENTARY1         PRESS_1       POLARITY+      Channel  1   Pin D25
Pedal  2        1   MOMENTARY1         PRESS_1       POLARITY+      Channel  1   Pin D26
Pedal  3        1   MOMENTARY1         PRESS_1       POLARITY+      Channel  1   Pin D27
Pedal  4        1   MOMENTARY1         PRESS_1       POLARITY+      Channel  1   Pin D12
Pedal  5        1   MOMENTARY1         PRESS_1       POLARITY+      Channel  1   Pin D13
Pedal  6        1   MOMENTARY1         PRESS_1       POLARITY+      Channel  1   Pin D17
Pedal  7        1   MOMENTARY1         PRESS_1       POLARITY+      Channel  1   Pin D35
Pedal  8      12L   MOMENTARY1         PRESS_1_2_L   POLARITY+      Channel  1   Pin D0
MTC None
... profile 1 loaded.

alf45tar, 2022-10-15:
Check you pedal configuration. You have 8 momentary1 and no ladder pedals.

Designbase, 2022-10-15:
With the Log above it was configured as Ladder...

Designbase, 2022-10-15:
Boot NORMAL
Loading profile 0 ...
Reading NVS Profile A ... done
Bank  2
Pedal  1        1   LADDER             PRESS_1       POLARITY+   CONTROL_CHANGE     100   Channel  1   Pin A36 D25
Pedal  2        1   LADDER             PRESS_1       POLARITY+   PROGRAM_CHANGE       2   Channel  2   Pin A37 D26
Pedal  3        1   LADDER             PRESS_1       POLARITY+   PROGRAM_CHANGE       3   Channel  3   Pin A38 D27
Pedal  4        1   LADDER             PRESS_1       POLARITY+   PROGRAM_CHANGE       0   Channel  4   Pin A39 D12
Pedal  5        1   LADDER             PRESS_1       POLARITY+   PROGRAM_CHANGE       0   Channel  5   Pin A32 D13
Pedal  6        1   LADDER             PRESS_1       POLARITY+   CONTROL_CHANGE      66   Channel  1   Pin A33 D17
Pedal  7        1   MOMENTARY1         PRESS_1       POLARITY+      Channel  1   Pin D35
Pedal  8      12L   MOMENTARY1         PRESS_1_2_L   POLARITY+      Channel  1   Pin D0

Designbase, 2022-10-15:
[image]

alf45tar, 2022-10-15:
show me the Actions

Designbase, 2022-10-15:
[image]
screenshot of actions...

alf45tar, 2022-10-15:
The second button of the the ladder is Pedal 1 Button 2 and not Pedal 2 Button 2.
If you attached a ladder to Pedal 1 you must create all the actions in Pedal 1.

Designbase, 2022-10-15:
Oh my God your are right! How stupid of me, now I see it I should have known....
Sorry to have bothered you... and thanks for supporting me...!
Rgds, Ron

closed 2022-10-15 by alf45tar