alf45tar/PedalinoMini #379: RGB order problem

bobvc133, 2022-10-16:
There is a problem with the RGB color order setting: change the color order in the options to GRB, although the LED can display red or green correctly, but setting the PC action and pressing the button repeatedly will flash green or red once. Send a MIDI signal to another PedalinoMini via IP MIDI, and the corresponding LED will display the opposite color. In addition, modifying the code of the Config.h file to rgbOrder = GRB has no effect, unless you modify the code of the Pedalino.h file to #define LED_RGB_ORDER GRB to solve the LED flashing wrong color, another PedalinoMini color shows the opposite problem.

alf45tar, 2022-10-16:
Check the latest 2.5.4

bobvc133, 2022-10-16:
After setting the color order to GRB in the option page, the LED can display the color normally, but modifying the factory setting rgbOrder = GRB in the Config.h code is still invalid, and must be modified in the option page.
[image]

bobvc133, 2022-10-16:
In addition, the documentation does not update the pictures and information about Show Incoming and Show Outcoming in the Interfaces page.

alf45tar, 2022-10-16:
> In addition, the documentation does not update the pictures and information about Show Incoming and Show Outcoming in the Interfaces page.

Noted. I will fix it soon or later. It is not on top of priorities.

alf45tar, 2022-10-16:
> After setting the color order to GRB in the option page, the LED can display the color normally, but modifying the factory setting rgbOrder = GRB in the Config.h code is still invalid, and must be modified in the option page.
> [image]

This change is not enough if not saved into NVS.

bobvc133, 2022-10-16:
I modified the code in the Config.h file EOrder order = GRB and the code in the Pedalino.h file EOrder rgbOrder = GRB solved my problem, but I don't know if there are other problems with this modification

alf45tar, 2022-10-16:
I do not think you will issues with this 2 changes.

closed 2022-10-16 by alf45tar