alf45tar/PedalinoMini #388: Display mode

bobvc133, 2022-11-10:
It is recommended to set a switch to turn off the action display, because most hardware modes cannot support it. For example, when I use a resistance ladder, the screen can only display the first button loading, and it is often incomplete.
[image]

bobvc133, 2022-11-11:
It is recommended to divide the Press action into two effects, one is the instantaneous effect and the other is the latch effect, so that the instantaneous button does not need to add the two actions of Press and Release, which can reduce the page layout space and reduce the difficulty of setting for users. After all, two actions have 4 values, 4 LED colors, and 4 labels.

alf45tar, 2022-11-11:
On momentary button the most used action should be PRESS triggered immediately on press. RELEASE is not su useful.
Replace PRESS with CLICK when you want to use DOUBLE PRESS or LONG PRESS because PRESS is triggered before every DOUBLE CLICK/LONG PRESS.

On latch button use PRESS and RELEASE or better I suggest to use CLICK. The CLICK behaviour on latch button is different. It corresponds to PRESS or RELEASE. Using CLICK you can reduce the number of duplicated actions.

Does it make sense?

bobvc133, 2022-11-11:
> On momentary button the most used action should be PRESS triggered immediately on press. RELEASE is not su useful. Replace PRESS with CLICK when you want to use DOUBLE PRESS or LONG PRESS because PRESS is triggered before every DOUBLE CLICK/LONG PRESS.
> 
> On latch button use PRESS and RELEASE or better I suggest to use CLICK. The CLICK behaviour on latch button is different. It corresponds to PRESS or RELEASE. Using CLICK you can reduce the numbers of duplicated actions.
> 
> Does it make sense?

Maybe I didn't explain it clearly. Let me give an example. First of all, my button hardware is all momentary buttons. If I want to add a latch effect, I only need to add a Press action, but to add a momentary effect, I must add a Press and A Release action, so that there will be 4 values, 4 LED colors and 4 labels. It takes many tests to know which value, color or label should be filled in to have the correct effect, so I recommend putting The Press action is divided into two effects, one is the instantaneous effect and the other is the latch effect.
[image]

[image]

alf45tar, 2022-12-15:
Press event will change soon. The Press latch mode will be replaced by Step by Step action (version 3.0.6).

bobvc133, 2022-12-19:
> Press event will change soon. The Press latch mode will be replaced by Step by Step action (version 3.0.6).

How to use the Step by Step action, I don’t understand, it seems to be different from what I expressed

alf45tar, 2022-12-19:
On every event the Step by Step action execute the next action in list from a sequence. Execution start from first item in the sequence and return to begin on first empty action (circular list of non empty actions).

alf45tar, 2022-12-20:
I cannot understand how you are planning to use the momentary switches.
A momentary switch has only one stable state. For normally open switches it is the RELEASE state. PRESS is not stable.

Do you want to keep it pressed by your foot until you need a RELEASE action? Or you press it and release immediately?

Can you explain your real scenario?

bobvc133, 2022-12-21:
> I cannot understand how you are planning to use the momentary switches. A momentary switch has only one stable state. For normally open switches it is the RELEASE state. PRESS is not stable.
> 
> Do you want to keep it pressed by your foot until you need a RELEASE action? Or you press it and release immediately?
> 
> Can you explain your real scenario? What should happen when you press it? With the current features a momentary switch can simulate a latch switch. A latch switch cannot simulate a momentary switch because to RELEASE it you have to press it.

What I have been talking about is the action configuration of the momentary switch. It has nothing to do with how the feet are operated. The current momentary effect action configuration has many duplicates and useless items, as shown in the blue and red marks in the picture below.
[image]

The suggestion I give is as shown in the figure below, so that the configuration of momentary effect actions is much simplified
[image]

alf45tar, 2022-12-21:
Now it is clear. You are asking to create a new action that combine Press and Release events in order to create a more short configuration.

This feature is similar to a keyboard key. Send Note On when pressed and Note Off when released. May I ask for what scenario you need it in a foot switch?

bobvc133, 2022-12-21:
> Now it is clear. You are asking to create a new action that combine Press and Release events in order to create a more short configuration.
> 
> This feature is similar to a keyboard key. Send Note On when pressed and Note Off when released. May I ask for what scenario you need it in a foot switch?

My purpose is to make the configuration easier. I often made mistakes when configuring transient effects before. Because of duplicate items and useless items, I had to test for a long time before I could get the correct transient effect configuration. In daily use, some equipment and software control needs Instantaneous effects, such as switching timbres, some need a signal to trigger, and some need two signals of 0 and 127 to trigger, the LED only needs to flash once, and does not need to stay on.

alf45tar, 2022-12-27:
Press&Release has been implemented and under testing. It will be released with next version 3.0.7.

bobvc133, 2022-12-27:
I found that just Press&Release still can't solve the similar transient effect problems I encountered, such as the action in Figure 2, using the Repeat Pressed event, the LED will remain on, and the correct state should be blinking.
[image]

bobvc133, 2022-12-27:
It is recommended to delete the RELEASE and LONG RELEASED events of PedalinoMini. They are of little significance to exist alone. Change other events (PRESS, CLICK, DOUBLE CLICK, LONG PRESS, REPEAT PRESSED) to default instantaneous effects, and add a HOLD MODE option Simulate the self-locking effect, you will find that all events can control the instantaneous effect or self-locking effect, and the LED status can also find the correct setting, especially the increment or decrement action (BANK SELECT+, BANK SELECT-, PROGRAM CHANGE+, PROGRAM CHANGE- , BANK+, BANK-, TAP, BPM+, BPM-) They should all have instantaneous effects, and each time they are triggered, the LED will flash once, not a constant light effect.