alf45tar/PedalinoMini #391: ABOUT PROFILES IN YOUTUBE VIDEO

bobvc133, 2022-11-12:
Can you post the full configuration files of both videos?
[image]

bobvc133, 2022-11-12:
Another question is is the encoder not suitable for the TTGO T DISPLAY development board?

alf45tar, 2022-11-12:
The full configuration of the first video is the 'mustang' configuration file. It is not yet updated for version 3 with Controls.
I do not have anymore the configuration of the second video because it was a test. Just set the Actions Send as Repeat.

Encoders can be used with any board.

bobvc133, 2022-11-12:
> The full configuration of the first video is the 'mustang' configuration file. It is not yet updated for version 3 with Controls. I do not have anymore the configuration of the second video because it was a test. Just set the Actions Send as Repeat.
> 
> Encoders can be used with any board.

There seems to be an error in the configuration file
[image]
[image]
[image]

alf45tar, 2022-11-12:
Use sponsor 2.5.4

bobvc133, 2022-11-12:
> Use sponsor 2.5.4

At the beginning of the test, the 2.5.4 firmware was used. I found that the configuration file format was not updated.

[image]
[image]

alf45tar, 2022-11-12:
Did you updated the Filesytem Image too?

bobvc133, 2022-11-12:
I always erase and re-burn

bobvc133, 2022-11-12:
> Did you updated the Filesytem Image too?

I uploaded the configuration file in the sponsor version, the action and pedal settings are normal, but on the TTGO development board, only the switch is working properly, the wheel is not normal, the EPS32 doit development board switch and the wheel are both normal

closed 2022-11-12 by bobvc133