alf45tar/PedalinoMini #395: Sequence action label setting problem

bobvc133, 2022-11-14:
“If action tag end with dot (.) the corresponding led and color is set on boot.”
When using the sequence, the action tags can be displayed normally, but the LED and color settings do not take effect
[image]

[image]
[image]

alf45tar, 2022-11-14:
Give a try to version 3.0.2

bobvc133, 2022-11-15:
> Give a try to version 3.0.2

The 3.0.2 firmware has other issues with the LEDs. When switching user profiles, the LEDs that were turned on in the previous user profile will remain on.

alf45tar, 2022-11-15:
Fixed. Version remain 3.0.2.

bobvc133, 2022-11-15:
> Fixed. Version remain 3.0.2.

Another new problem was found, the LED will flash the set color first during the network connection process.

alf45tar, 2022-11-15:
This is not a bug. PedalinoMini works via Bluetooth during WiFI connection (or if WiFi is disabled).
The initial led status is set before WiFi connection.

bobvc133, 2022-11-17:
> Give a try to version 3.0.2

Could you please fix the problem with the 2.5.4 firmware? Because I use the resistance ladder mode and have many buttons, the control function of version 3.0.2 is not very practical for me, and it is very inconvenient to configure.

alf45tar, 2022-11-20:
Sorry but I have no time to back port fixes to old versions. I am supporting only the last one.
However the issue can be fixed removing the -1 from the following lines

https://github.com/alf45tar/PedalinoMini/blob/009cb836cf7acb08430078b100e7f4c5ac19dbd0/src/Controller.h#L1717-L1718

closed 2022-11-20 by alf45tar

bobvc133, 2023-01-08:
3.0.7 firmware, sending Step by Step has reproduced this problem, and the action tag setting does not take effect.