alf45tar/PedalinoMini #399: Long Press action not triggered if button is part of a multi-pedal control

potul, 2022-11-15:
If one button is present in two controls, one only with itself, and the other with another button (for multi-button actions), Long press action for the first control is not triggering an action.

Event is detected by the monitor, but the action linked to it does not trigger.

Repeat Pressed and Long Released work fine, it's just Long Press that does not work.

Deleting the multi-button control fixes the problem, so it seems directly linked to this.

potul, 2022-11-15:
I forgot to mention. The issue only appears if the multi-button control is used somewhere in another action.

alf45tar, 2022-11-15:
Increase the Repeat Press Time under Options at 570ms.
The difference between Long Press and Repeat Press Time must be greater than PED_SIMULTANEOUS_GAP (50ms).
I will investigate if this limit can be removed.

alf45tar, 2022-11-20:
Added Debounce Interval and Simultaneous Gap Time in Options.

alf45tar, 2022-11-27:
Rewritten simultaneous press detection and removed the limitation.

potul, 2022-12-12:
Ok I tested it today, and still buttons are behaving strangely when using double buttons. If not, everything seems to work fine. One example: When using a Control with 2 buttons and pressing both, the press is detected fine, but then the release of the individual buttons is also triggered.

alf45tar, 2022-12-13:
Do you think that release event after a simultaneous press should not be triggered?

potul, 2022-12-13:
> Do you think that release event after a simultaneous press should not be triggered?

I was expecting the double button control release event to be triggered, but not the individual buttons.
From my perspective, as soon as a simultaneos press is detected, the individual button events should be disabled (press, long press, release, double click, click) until the double press is released, because I want to treat it like it was a third stompswitch. Think about the use case as an example:
-I want button 1 to send a momentaneous CC message with value 127, so I set it so that when it's released it sends a CC with value 0. 
-Button 2 to do something else.
-And I want that if I press both buttons, a PC message is sent. 
I don't want the CC with 0 value to be sent when I'm pressing both buttons.

In some cases it might not hurt to have the individual release event triggered, in others it might. And you can always add whatever action to the release event of the double button control if you need to.

potul, 2022-12-13:
And sorry, I think I commented the wrong issue. This one can be closed as the specific problem was resolved, but the other one I opened and is still open is the one I wanted to comment on. #400

closed 2022-12-13 by alf45tar