alf45tar/PedalinoMini #400: Release event not detected

potul, 2022-11-16:
Sponsor version 3.0.2

I've had  some cases when Release event was not detected when double-clicking or long pressing.

After some testing I have found that even if double-click is not enabled, if I click twice the same button with a time between presses of approx the "Double Press Time" config, the release is not detected. See log below:

Pedal: 2     Button: 1    EventType: 0     ButtonState: 0
Action: UND/RED
CONTROL CHANGE.....Code  63......Value 127.....Channel  1
Pedal: 2     Button: 1    EventType: 2     ButtonState: 0
Pedal: 2     Button: 1    EventType: 2     ButtonState: 1

Press is detected, and then 2 clicks, but no release. Changing the "Double Press Time" changes the spacing of presses that cause the issue, but the issue remains.

A similar behaviour happens if I long click and release at certain timings:

Pedal: 2     Button: 1    EventType: 0     ButtonState: 0
Action: UND/RED
CONTROL CHANGE.....Code  63......Value 127.....Channel  1
Pedal: 2     Button: 1    EventType: 4     ButtonState: 0
Pedal: 2     Button: 1    EventType: 5     ButtonState: 0
Pedal: 2     Button: 1    EventType: 5     ButtonState: 0

In this case, press, long press and repeat are detected, but no release, nor long release.

potul, 2022-11-16:
Let me add that I just realized this issue is only happening when the button is part of a two-button control, and the two-button control has some action assigned to it.
So it seems to be related to my previous issue, and linked to timing. If this can be resolved via configuration, let me know.

alf45tar, 2022-12-10:
Did you try with the latest version if solved?

potul, 2022-12-12:
No sorry. As the sponsor version was giving me a lot of issues, I reinstalled the public one, giving up on the double button capability.
As soon as I can I will test it and let you know

alf45tar, 2022-12-13:
There are at least 2 mode to solve it.

1. Define an action for simultaneous control on release event. In this case the single release events are not triggered.

2. Do not use the release event. Use only press. The event linked to press and release can be moved to a sequence and execute the sequence Step by Step (it is a new feature within the next version 3.0.6). On every press you can execute a different action from a list.

potul, 2022-12-13:
Great, I will give it a try. The new sequence step by step is a nice addition, and will cover some  gaps in funcionality. Good to know.

alf45tar, 2023-01-08:
reopen it if you think it is not solved

closed 2023-01-08 by alf45tar