alf45tar/PedalinoMini #401: Synchronization of LED status

bobvc133, 2022-11-17:
When PedalinoMIni receives from any supported interface a NoteOn/CC/PC midi message corresponding to a pedal it turn on (or off) the corresponding led.

I used two PEDALINOMINI to test and found that only when receiving PC signal, the corresponding LED will be turned on or off. When receiving CC and NOTE signal, only the screen will display the received MIDI signal.

alf45tar, 2022-11-20:
The circuit is working fine but it requires a lot of changes in the software with no value. If you need more buttons increase the LADDER_STEPS and recalculate the resistors.

alf45tar, 2022-11-20:
Check version 3.0.3 for led synchronisation fix

AndreySom, 2022-11-28:
First, I'd like to thank you @alf45tar  for such a nice project. I've built mine and it works like a charm. I became a sponsor today, and I had the same led sync question as bob posted above. I'd like to know where can i find version 3 to test it. I'm using IPMidi.

alf45tar, 2022-11-28:
@AndreySom I added you to the private repository. You should have received an email.

alf45tar, 2022-11-28:
Very strange. I didn't change too much in the WebUI this time. It seems it is mix of 2 pages: Controls and Options.

alf45tar, 2022-12-04:
Sill an issue?

bobvc133, 2022-12-05:
> Sill an issue?

3.0.4 firmware, when editing the transient effect, the PRESS and RELEASE actions can still only set the OFF color, so the LED brightness is always OFF brightness, and it is strange when editing the NOTE ON and NOTE OFF actions, the ON and OFF colors of the RELEASE action can be set setting, but the ON color cannot be synchronized, and the option page encoder layout problem still exists

alf45tar, 2022-12-10:
I am not able to reproduce the Options page layout issue. It works fine for me and I have no evidence from others.
Did you try to cancel the cache of the browser? Did you uploaded the new filesystem image? Did you try to fully erase before flash the new firmware?

bobvc133, 2022-12-11:
> I am not able to reproduce the Options page layout issue. It works fine for me and I have no evidence from others. Did you try to cancel the cache of the browser? Did you uploaded the new filesystem image? Did you try to fully erase before flash the new firmware?

1. Every time I burn the firmware, I will erase it, and then recompile and upload the firmware and file system image. My browser has always disabled the caching function. The test uses the original firmware and only modifies the port.
[image]
[image]
[image]3
[image]
2. There are still problems with the layout of the encoder bar on the 3.0.5 firmware option page in various resolutions
3. There are still problems with the LED setting and status synchronization of the instantaneous action of the 3.0.5 firmware
4. I also found a problem that the CC action of the instant effect can only be set in TAG1, and the NOTE action can only be set in TAG2.
[image]
[image]

alf45tar, 2022-12-11:
Encoders tile when on single column fixed in version 3.0.6.
Tag1 and Tag2 are used as is by design. It is not considered a bug. May be in some next version I will create anything more coherent.
Can you be more specific on point 3?

bobvc133, 2022-12-11:
> Encoders tile when on single column fixed in version 3.0.6. Tag1 and Tag2 are used as is by design. It is not considered a bug. May be in some next version I will create anything more coherent. Can you be more specific on point 3?

First of all, set the press and release actions at the same time, the ON color setting of the press is invalid, only the OFF color can be set, so the LED brightness can only be OFF brightness. The Note Off Release action can set the ON color, but CC cannot. In the end, the LED status cannot be synchronized. These problems all occur in the momentary effect.
[image]
[image]

closed 2023-01-07 by bobvc133