alf45tar/PedalinoMini #406: 3 ladders unexpected led behaviour

jimhiggs, 2022-12-07:
This is my second build underway and I have just updated to 2.5.2
I have 3 ladders of 6 switches each and each has an led so I have changed led in config to 18.
I have configured each pedal with 6 actions each with an id tag so pedal 1/1, pedal 1/2, pedal 3/5 etc and all buttons are firing the correct press functions an displaying the correct tags. Problem is with leds. I have tried numbering them at pedal level and at action level and get the same results. When I press pedal 1/1, the leds for pedal 1/1, pedal 2/1 and pedal 3/1 all light using the colors defined for each action, same if I press pedal 2/5, the leds for 1/5, 2/5 and 3/5 all light.
I know the led chain is working correctly because I have tried setting the led number for pedal 1/1 to each number 1-18 and this works.  Any ideas?

Another thing I can't wrap my head round is that with the ladder if I press button 1 it changes to the on color and remaining five show off color. If I press button 2 it switches to the on color and switches button 1 color to the off color. Like its an or situation. How can I configure it so pressing button 1 switches it to on color and pressing other buttons on the same ladder do not turn it off but pressing the same button again does?  Simulating 6 independent latch switches I suppose?

alf45tar, 2022-12-07:
Can you share you configuration file?

jimhiggs, 2022-12-07:
I used Github to load 2.5.2 and only changed #define LEDS 18 in pedalino.h.  Rest is untouched.  
Here is my Configuration file.
[Dec 22B.zip](https://github.com/alf45tar/PedalinoMini/files/10177081/Dec.22B.zip)

alf45tar, 2022-12-07:
Only one Program Change can be active on each MIDI channel. A new Program Change deletes the last one (on the same channel) activating the off color (green in you config for ladder 1) of other button.
I suggest to use a Control Change.

jimhiggs, 2022-12-07:
OK Thanks. I did not know that, will try that. I have just sponsored and will switch to the latest 3.0.4. do you want me to keep this version going in case you want further feedback on this issue?

alf45tar, 2022-12-07:
If you want to share some photos of your build I will add one to the homepage.