alf45tar/PedalinoMini #408: Action setting problem

bobvc133, 2022-12-11:
Can the options in the red box be used normally? The STAR and STOP actions can only be set to ON, BUTTON, and SEND items, and the OLED display also has problems.
[image]
[image]

alf45tar, 2022-12-11:
Start, Stop and Continue are MIDI messages related to MIDI Clock. The OLED display will be fixed with the next version.
MTC Start, MTC Stop, MTC Continue, Tap, BPM+ and BPM- are related to Midi Time Code.

MIDI Clock and MTC are already implemented in the code but they is no user interface to activate/operate yet.

bobvc133, 2022-12-12:
When will the MIDI clock and MTC function be able to be used normally? This is also a relatively important function.

closed 2023-01-07 by bobvc133