alf45tar/PedalinoMini #413: 3.0.7 Summary of Firmware Problems

bobvc133, 2023-01-06:
1. Create a new configuration file, the configuration editor will prompt invalid attributes, multiple events will also be confused, and the Press & Release event will be saved as NONE
[image]
[image]

2. After editing the off color of the Release event LED and saving it, the on color will automatically be the same as the off color.
[image]
[image]

3. It is completely impossible to set the latch effect. It is recommended to consider the solution I proposed in #388.
https://github.com/alf45tar/PedalinoMini/issues/388#issuecomment-1366016368
[image]
[image]

bobvc133, 2023-01-06:
A little later, I will illustrate the third point of problem improvement suggestions in the form of pictures

bobvc133, 2023-01-06:
With the current features a momentary switch can simulate a latch switch. A latch switch cannot simulate a momentary switch because to RELEASE it you have to press it，Therefore, all functions can be perfectly realized only by using momentary switches. My following instructions are based on the premise that the hardware uses momentary switches.

In daily use, some equipment and software controls require instantaneous effects, not simple on and off effects.
1. For example, to switch timbre or bank, sometimes a 0 or 127 signal is needed to trigger, and sometimes two signals of 0 and 127 are needed to trigger at the same time. Although the Press+Release event and the newly added Press&Release event can achieve instantaneous effects, they cannot achieve only Send a 0 or 127 signal, always send the signal twice, possibly switch two timbres or BNAK
[image]

But other events can't realize instantaneous effect, and can't send two signals of 0 and 127 at the same time.
[image]

alf45tar, 2023-01-06:
If you want to send multiple MIDI messages for one event use Sequence.

bobvc133, 2023-01-06:
> If you want to send multiple MIDI messages for one event use Sequence.

The pictures are just for the convenience of taking screenshots, so they are all displayed on one control. It is not to send multiple signals that may conflict on one control. The description of the third point is still being revised

bobvc133, 2023-01-06:
> If you want to send multiple MIDI messages for one event use Sequence.

Regarding the questions and suggestions about the instant effect, although I have explained it many times in several issues and comments, due to translation problems and cultural differences, I may still not express it clearly. After a variety of devices, software and usage scenarios After repeated testing and referring to the conventional settings of similar MIDI pedals and software, I think that the functions of PedalinoMini can be optimized to be more powerful, and the settings are more abundant and free.The purpose of raising so many issues is just to help test and improve PedalinoMini better. Of course, this is just my personal experience and suggestion, rather than asking to improve PedalinoMini according to my personal ideas. This is the last release about transient effects issues, I don't want to be your burden.

alf45tar, 2023-01-07:
Don't worry. I appreciate any suggestion. 
Regarding the "hold mode" I think it is already implemented in Step by Step action apart for Bank+ and Bank-.

bobvc133, 2023-01-07:
> Don't worry. I appreciate any suggestion. Regarding the "hold mode" I think it is already implemented in Step by Step action apart for Bank+ and Bank-.

Can you give an example of the configuration? The requirement is to need two buttons, one to send BANK+, one to send BANK-, the LED lights up and sends the value when pressed, and the LED turns off when released.

alf45tar, 2023-01-07:
At the moment it is not possible because Bank+/Bank- cannot be used in Sequences.

bobvc133, 2023-01-07:
> At the moment it is not possible because Bank+/Bank- cannot be used in Sequences.

That's why I have been emphasizing that all events of the momentary switch should default to momentary effects. You can only send one value, then leave a value blank, and add a HOLD MODE function, allowing users to choose whether to use momentary effects or latches according to their needs Effect. because the above Bank+/Bank- requirements cannot be perfectly realized by any event.

bobvc133, 2023-01-07:
For example, the effect software TH-U only needs a value of 0 or 127 to switch the timbre bank or timbre, but PedalinoMini still cannot be perfectly realized. The Press+Release event and the newly added Press&Release event send two values at a time, and switch the timbre twice. Libraries and timbres, although other events can send a value, but the LED will remain on when pressed once.
[image]
[image]
[image]

alf45tar, 2023-01-07:
The logic of Press&Release is different.
Press&Release do not send 2 values at a time. It send one value on Press and one value on Release. One value for each event because between Press and Release can pass any amount of time. It has been created to aggregate 2 events in one action but it do not add any new feature.

It looks like that TH-U is fully configurable. I see the Learn button. Why you are not configuring the Press event to one button for "Previous bank", another one for "Next bank", another one for "Previous preset" and another one for "Next preset"?

bobvc133, 2023-01-07:
Learning to operate is not a problem, but the LED will remain on after the press event is pressed and cannot be turned off. You can test this software

alf45tar, 2023-01-07:
Do not use led for these type of actions where it doesn't make sense.
At the moment the led flashing is not supported.
Forget Press&Release. It is not implemented to flashing leds.

bobvc133, 2023-01-07:
I use a lot of MIDI pedals, don't know why the regular settings don't make sense
[image]

https://user-images.githubusercontent.com/91867813/211164304-67e76608-6e1f-4ff6-9918-096a94754f46.mp4

bobvc133, 2023-01-07:
[image]

https://user-images.githubusercontent.com/91867813/211164717-248e9025-7345-4e6f-8b6d-3474780d80a9.mp4

bobvc133, 2023-01-07:
https://user-images.githubusercontent.com/91867813/211165328-72553c40-653c-4ab2-80ec-93f6c86f80d5.mp4

alf45tar, 2023-01-07:
I mean a fixed led color has less sense than other scenario and flashing led for press confirmation is not supported.

bobvc133, 2023-01-07:
Well, I close this question, because of translation errors and cultural differences, this question may not be able to communicate clearly

closed 2023-01-07 by bobvc133