alf45tar/PedalinoMini #420: can't build and upload

WernAir77, 2023-01-20:
Hi, I have a working pedalino with a very old software version. I build it 2020 or so. I can't update OTA, so I tried to flash the ESP with the newest version. Build after choosing build about 17 errors exit and after trying to upload nothing happened, but error 1.
I know I had similar issues at the first time, but I can't remember how I solved them.
I run Ventura on a M1 Mac. I downloaded VSCode and Xcode, installed platformIO, performed git: clone, select the SSH display in line 14/15 and select Esp32doit-devkit-v1 in the project tasks, then build and after that upload or upload an Monitor.

Problems during building:
[{
	"resource": "/Users/wernair/PedalinoMini-master/.pio/libdeps/esp32doit-devkit-v1/AppleMIDI/src/rtpMIDI_Parser_JournalSection.hpp",
	"owner": "cpp",
	"severity": 4,
	"message": "unused variable 'S_flag' [-Wunused-variable]",
	"startLineNumber": 140,
	"startColumn": 18,
	"endLineNumber": 140,
	"endColumn": 18
},{
	"resource": "/Users/wernair/PedalinoMini-master/.pio/libdeps/esp32doit-devkit-v1/AppleMIDI/src/rtpMIDI_Parser_JournalSection.hpp",
	"owner": "cpp",
	"severity": 4,
	"message": "unused variable 'channelNr' [-Wunused-variable]",
	"startLineNumber": 141,
	"startColumn": 21,
	"endLineNumber": 141,
	"endColumn": 21
},{
	"resource": "/Users/wernair/PedalinoMini-master/.pio/libdeps/esp32doit-devkit-v1/AppleMIDI/src/rtpMIDI_Parser_JournalSection.hpp",
	"owner": "cpp",
	"severity": 4,
	"message": "unused variable 'H_flag' [-Wunused-variable]",
	"startLineNumber": 142,
	"startColumn": 18,
	"endLineNumber": 142,
	"endColumn": 18
},{
	"resource": "/Users/wernair/PedalinoMini-master/.pio/libdeps/esp32doit-devkit-v1/BLE-MIDI/src/BLEMIDI_Transport.h",
	"owner": "cpp",
	"severity": 4,
	"message": "unused variable 'timestampHigh' [-Wunused-variable]",
	"startLineNumber": 260,
	"startColumn": 14,
	"endLineNumber": 260,
	"endColumn": 14
},{
	"resource": "/Users/wernair/PedalinoMini-master/.pio/libdeps/esp32doit-devkit-v1/BLE-MIDI/src/BLEMIDI_Transport.h",
	"owner": "cpp",
	"severity": 4,
	"message": "variable 'timestamp' set but not used [-Wunused-but-set-variable]",
	"startLineNumber": 263,
	"startColumn": 18,
	"endLineNumber": 263,
	"endColumn": 18
},{
	"resource": "/Users/wernair/PedalinoMini-master/.pio/libdeps/esp32doit-devkit-v1/TFT_eSPI/TFT_eSPI.h",
	"owner": "cpp",
	"severity": 4,
	"message": "#warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]",
	"startLineNumber": 955,
	"startColumn": 8,
	"endLineNumber": 955,
	"endColumn": 8
},{
	"resource": "/Users/wernair/PedalinoMini-master/src/Controller.h",
	"owner": "cpp",
	"severity": 4,
	"message": "'void adc_power_off()' is deprecated [-Wdeprecated-declarations]",
	"startLineNumber": 1366,
	"startColumn": 15,
	"endLineNumber": 1366,
	"endColumn": 15
},{
	"resource": "/Users/wernair/PedalinoMini-master/src/Controller.h",
	"owner": "cpp",
	"severity": 4,
	"message": "'void adc_power_off()' is deprecated [-Wdeprecated-declarations]",
	"startLineNumber": 1366,
	"startColumn": 29,
	"endLineNumber": 1366,
	"endColumn": 29
},{
	"resource": "/Users/wernair/PedalinoMini-master/src/LedsEffects.h",
	"owner": "cpp",
	"severity": 4,
	"message": "variable 'thissat' set but not used [-Wunused-but-set-variable]",
	"startLineNumber": 311,
	"startColumn": 20,
	"endLineNumber": 311,
	"endColumn": 20
},{
	"resource": "/Users/wernair/PedalinoMini-master/src/Pedalino.h",
	"owner": "cpp",
	"severity": 4,
	"message": "format '%d' expects argument of type 'int', but argument 4 has type 'long int' [-Wformat=]",
	"startLineNumber": 508,
	"startColumn": 57,
	"endLineNumber": 508,
	"endColumn": 57
},{
	"resource": "/Users/wernair/PedalinoMini-master/src/UdpMidiOut.h",
	"owner": "cpp",
	"severity": 4,
	"message": "'midi::MidiInterface<Transport, Settings, Platform>& midi::MidiInterface<Transport, _Settings, _Platform>::sendPolyPressure(midi::DataByte, midi::DataByte, midi::Channel) [with Transport = appleMidi::AppleMIDISession<WiFiUDP>; _Settings = appleMidi::AppleMIDISettings; _Platform = midi::DefaultPlatform; midi::DataByte = unsigned char; midi::Channel = unsigned char]' is deprecated [-Wdeprecated-declarations]",
	"startLineNumber": 126,
	"startColumn": 104,
	"endLineNumber": 126,
	"endColumn": 104
},{
	"resource": "/Users/wernair/PedalinoMini-master/src/UdpMidiOut.h",
	"owner": "cpp",
	"severity": 4,
	"message": "'midi::MidiInterface<Transport, Settings, Platform>& midi::MidiInterface<Transport, _Settings, _Platform>::sendPolyPressure(midi::DataByte, midi::DataByte, midi::Channel) [with Transport = ipMidi::ipMidiTransport<WiFiUDP>; _Settings = midi::DefaultSettings; _Platform = midi::DefaultPlatform; midi::DataByte = unsigned char; midi::Channel = unsigned char]' is deprecated [-Wdeprecated-declarations]",
	"startLineNumber": 253,
	"startColumn": 102,
	"endLineNumber": 253,
	"endColumn": 102
},{
	"resource": "/Users/wernair/PedalinoMini-master/src/WebConfigAsync.h",
	"owner": "cpp",
	"severity": 4,
	"message": "suggest parentheses around '&&' within '||' [-Wparentheses]",
	"startLineNumber": 5055,
	"startColumn": 53,
	"endLineNumber": 5055,
	"endColumn": 53
},{
	"resource": "/Users/wernair/PedalinoMini-master/src/WebConfigAsync.h",
	"owner": "cpp",
	"severity": 4,
	"message": "'void adc_power_off()' is deprecated [-Wdeprecated-declarations]",
	"startLineNumber": 5127,
	"startColumn": 5,
	"endLineNumber": 5127,
	"endColumn": 5
},{
	"resource": "/Users/wernair/PedalinoMini-master/src/WebConfigAsync.h",
	"owner": "cpp",
	"severity": 4,
	"message": "'void adc_power_off()' is deprecated [-Wdeprecated-declarations]",
	"startLineNumber": 5127,
	"startColumn": 19,
	"endLineNumber": 5127,
	"endColumn": 19
},{
	"resource": "/Users/wernair/PedalinoMini-master/Users/wernair/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c",
	"owner": "cpp",
	"severity": 4,
	"message": "initialization from incompatible pointer type [-Wincompatible-pointer-types]",
	"startLineNumber": 922,
	"startColumn": 39,
	"endLineNumber": 922,
	"endColumn": 39
},{
	"resource": "/Users/wernair/PedalinoMini-master/Users/wernair/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c",
	"owner": "cpp",
	"severity": 4,
	"message": "initialization from incompatible pointer type [-Wincompatible-pointer-types]",
	"startLineNumber": 923,
	"startColumn": 40,
	"endLineNumber": 923,
	"endColumn": 40
}]

Problems during upload:
Traceback (most recent call last):
  File "/Users/wernair/.platformio/penv/lib/python3.9/site-packages/serial/serialposix.py", line 322, in open
Serial port /dev/tty.SLAB_USBtoUART
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/tty.SLAB_USBtoUART'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/wernair/.platformio/packages/tool-esptoolpy/esptool.py", line 4582, in <module>
    _main()
  File "/Users/wernair/.platformio/packages/tool-esptoolpy/esptool.py", line 4575, in _main
    main()
  File "/Users/wernair/.platformio/packages/tool-esptoolpy/esptool.py", line 4074, in main
    esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts,
  File "/Users/wernair/.platformio/packages/tool-esptoolpy/esptool.py", line 120, in get_default_connected_device
    _esp = chip_class(each_port, initial_baud, trace)
  File "/Users/wernair/.platformio/packages/tool-esptoolpy/esptool.py", line 313, in __init__
    self._port = serial.serial_for_url(port)
  File "/Users/wernair/.platformio/penv/lib/python3.9/site-packages/serial/__init__.py", line 90, in serial_for_url
    instance.open()
  File "/Users/wernair/.platformio/penv/lib/python3.9/site-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/tty.SLAB_USBtoUART: [Errno 2] No such file or directory: '/dev/tty.SLAB_USBtoUART'
*** [upload] Error 1
===================================== [FAILED] Took 4.71 seconds =====================================

Environment          Status    Duration
-------------------  --------  ------------
esp32doit-devkit-v1  FAILED    00:00:04.714

I need help...

alf45tar, 2023-01-20:
`serial.serialutil.SerialException: [Errno 2] could not open port /dev/tty.SLAB_USBtoUART: [Errno 2] No such file or directory: '/dev/tty.SLAB_USBtoUART'`

I think you didn't installed the driver to use the serial via USB. Check platformio.ini for more info or download them from 

https://www.silabs.com/documents/public/software/Mac_OSX_VCP_Driver.zip

Regards
alf45tar

WernAir77, 2023-01-20:
Thanks a lot. Problem is solved!

closed 2023-01-21 by alf45tar