alf45tar/PedalinoMini #421: TTGO T display restarts repeatedly

bobvc133, 2023-01-23:
Pedal 4 using the voltage ladder will result in a black screen and repeated restarts and cannot start normally. Click any button on Pedal 4 or remove the pull-up resistor of PIN_A4 to start normally. I checked and it seems to be a problem with the Strapping pin.
[image]
[image]

alf45tar, 2023-01-23:
PIN_A4 is GPIO39 and apart it is a input pin only there is no restriction on boot.

alf45tar, 2023-01-23:
It looks like that burning some efuse can solve it but we need to read more documentation.

bobvc133, 2023-01-23:
> PIN_A4 is GPIO39 and apart it is a input pin only there is no restriction on boot. PIN_D4 is GPIO12 must be LOW during boot. Never had issues but I will check again.

I tested and found that only the pedal 4 will restart continuously when using the voltage ladder, which should be caused by the level problem of the GPIO12 pin.

bobvc133, 2023-02-03:
> 好像烧一些efuse就可以解决，但是我们需要阅读更多的文档。
> 
> `GPIO12 (MTDI) is used as a bootstrapping pin to select output voltage of an internal regulator which powers the flash chip (VDD_SDIO). This pin has an internal pulldown so if left unconnected it will read low at reset (selecting default 3.3V operation). When adding a pullup to this pin for SD card operation, consider the following: For boards which don't use the internal regulator (VDD_SDIO) to power the flash, GPIO12 can be pulled high. For boards which use 1.8V flash chip (e.g. WROVER), GPIO12 needs to be pulled high at reset. This is fully compatible with SD card operation. On boards which use the internal regulator and a 3.3V flash chip (e.g. ESP-WROOM-32), GPIO12 must be low at reset. This is incompatible with SD card operation. In most cases, external pullup can be omitted and an internal pullup can be enabled using a gpio_pullup_en(GPIO_NUM_12); call. This is handled by SD Card driver. Another option is to burn the flash voltage selection efuses. This will permanently select 3.3V output voltage for the internal regulator, and GPIO12 will not be used as a bootstrapping pin. Then it is safe to connect a pullup resistor to GPIO12. This option is suggested for production use. esp-idf_path>/components/esptool_py/esptool/espefuse.py set_flash_voltage 3.3V`

I replaced the other voltage ladders and avoided this problem. It is recommended to delete the problematic voltage ladder, the function is the same, and a digital interface can also be saved.

[image]