alf45tar/PedalinoMini #422: ESP-32-S3 Development Board

bobvc133, 2023-01-26:
I bought an ESP32-S3 development board, USB supports CDC-ACM virtual serial port and JTAG transfer function, can realize USB MIDI function, and supports lithium battery charging and discharging, VIN port charging and power supply, power detection and other functions, and more I feel that this development board is more suitable for PedalinoMini. There are basically no shortcomings in all aspects. The price on AliExpress is only $5.26. It is recommended to consider adapting the ESP32-S3 core, because some development boards currently supported by PedalinoMini either have insufficient GPIOs, or have been discontinued or will soon be discontinued, and the prices that can be purchased are not cheap.
[image]
https://www.aliexpress.com/item/1005004428945296.html?spm=5261.ProductManageOnline.0.0.48af4edfYbyEoI
https://wiki.banana-pi.org/BPI-Leaf-S3

alf45tar, 2023-01-26:
It is interesting but there are pro and cons.

Pros

- USB MIDI
- A lot of GPIO

Cons

- No display
- It not cheaper than TTGO T-Display board

bobvc133, 2023-01-26:
> It is interesting but there are pros and cons.
> 
> Pros
> 
> * USB MIDI
> * A lot of GPIOs
> 
> Cons
> 
> * No display
> * It not cheaper than TTGO T-Display board

1. I bought a lot of monitors compatible with SSD1306 drivers, 0.96, 1.3, and 2.42 inches, and the price is not expensive. The point is to be more selective, and a separate module is more convenient for installation and layout.
2. There is no need to purchase an additional Arduino Pro Micro when using USB MIDI.
3. The development boards other than the TTGO T-Display board cannot perfectly realize the functions of battery power charging and power detection.
[image]
[image]
[image]

bobvc133, 2023-01-27:
> That board looks interesting, because the boot- button can also be reached using a pin. I'd still like to know where to get the right connectors for the battery and power.

The battery can use 18650 batteries and other 3.7V batteries, and the battery connector terminal model is JST 1.25

rigr, 2023-01-27:
Thank you - just found a pdf and there the connector is calles "ph-2awt". The sda-connector is called "J1 Conn_01x04_Female"
Do you know where to source the connectors? Specially the one for the battery would be good to have. 
[BPI-Leaf-S3-Chip-V0.1A.pdf](https://github.com/alf45tar/PedalinoMini/files/10516878/BPI-Leaf-S3-Chip-V0.1A.pdf)

bobvc133, 2023-01-27:
> Thank you - just found a pdf and there the connector is calles "ph-2awt". The sda-connector is called "J1 Conn_01x04_Female" Do you know where to source the connectors? Specially the one for the [...]

Just search for keywords, there are many on AliExpress, and most of them are made in China
[image]

rigr, 2023-02-01:
Thankyou. Unfortunately neither the banana pi leaf nor the wires can be delivered to here for a reasonable price. :(

bobvc133, 2023-02-03:
> Thankyou. Unfortunately neither the banana pi leaf nor the wires can be delivered to here for a reasonable price. :(

ESP32 S3 has not been on the market for a long time, and the various platforms should not arrive. It will be fine after a while. If you can buy TTGO and other development boards, you should also be able to buy Banana Pi. After all, most development boards are produced in China.

rigr, 2023-02-12:
A handful of boards come directly from the producer and I wait for delivery. I'm curious. :)

bobvc133, 2023-02-13:
> A handful of boards come directly from the producer and I wait for delivery. I'm curious. :)

 banana pi？

Sthopeless, 2023-02-14:
[image]

I don't think is that expensive vs any other of PedalinoMini options. 
I think if we factor it doesn't need the Arduino and it offers a lot of pins, is extremely good option.
I don't mind not having a display since I already use the I2C.

The "ESP32 S2 Mini" also looks a very good and extremely cheap alternative.

[image]

bobvc133, 2023-02-14:
> [image]
>
> I don't think is that expensive vs any other of PedalinoMini options. I think if we factor it doesn't need the Arduino and it offers a lot of pins, is extremely good option. I don't mind [...]

If you don’t need the battery and power detection function, then ESP32 S2 Mini is indeed a more economical choice, but I found that many people still find ways to add external battery and power detection components. After comparison, I think banana pi BPI-Leaf-S3 It should be the best solution. It can basically meet all the needs. It is very suitable for PedalinoMini. The only problem is that it does not support IPEX antennas and cannot use metal casings. However, this can also be solved by DIY external wires.

alf45tar, 2023-02-15:
I ordered a Banana Pi Leaf S3. It could be supported in the near future.

bobvc133, 2023-02-15:
> I ordered a Banana Pi Leaf S3. It could be supported in the near future.

Great, the future PedalinoMini will become more and more powerful.

rigr, 2023-02-17:
@bobvc133 - you wrote the pi-leaf-S3 board "can realize USB MIDI function" - what changes have to be done to the libraries?
It looks like the MIDIUSB.h library does not (yet) know that this board supports USB MIDI:

Arduino-compiler errors:
MIDIUSB.h:18:2: error: #error MIDIUSB can only be used with an USB MCU.
 #error MIDIUSB can only be used with an USB MCU.
and
MIDIUSB.h:78:2: error: #error "Unsupported architecture"
 #error "Unsupported architecture"

bobvc133, 2023-02-21:
> @bobvc133 - you wrote the pi-leaf-S3 board "can realize USB MIDI function" - what changes have to be done to the libraries? It looks like the MIDIUSB.h library does not (yet) know that this board [...]

You can look up relevant information, I'm busy with other things

rigr, 2023-02-21:
- ok. 
The MIDIUSB.h github also does not (yet) support the ESP32 S2 and S3, so we need to find our own solution or wait.

Sthopeless, 2023-02-23:
> @bobvc133 - you wrote the pi-leaf-S3 board "can realize USB MIDI function" - what changes have to be done to the libraries? It looks like the MIDIUSB.h library does not (yet) know that this board [...]

You are trying to flash the "USB MIDI Converter" example into the Leaf S3 which is not the goal, for that you probably will need to search for another library compatible with ESP32 S3.
I have received my board yesterday, a lot bigger than I thought it would be :smile: 

Best wait for @alf45tar input since being compatible and having "USB to MIDI converter" feature are 2 different things

rigr, 2023-02-23:
I searched on the bpi forum and the github pages of MIDIUSB and the existing library does not (yet) support the esp32 S3. So sure I wait and have a look how it will be done but I'm sure, some library or parts of it will be used. And I asked bobvc133, because I thought maybe he knows. (I'm currently working on a different project using the bpi-leaf-s3 and would be happy to have MIDI over USB there.)

jgilbertfpv, 2023-02-24:
www.amazon.com/LILYGO-T-Display-S3-ESP32-S3-Development-Soldered/dp/B0BF542H39/
Just came across this post and seen this last night as an option with a larger screen and faster refresh rate. been using the regular lilygo-t-display version for a while now and it's been working fine, majority of the pins seem to match up.

Sthopeless, 2023-02-24:
> After first experiment with the board I think it is not yet fully supported and stable. For example I cannot connect to wifi [espressif/arduino-esp32#7869](https://github.com/espressif/arduino-esp32/issues/7869)

@alf45tar I ran the example provided in your url using Arduino IDE and my Leaf connected to the WiFi.
[image]

I have only done some testings using circuit python but I have tested sending USB HID keyboard and MIDI keys and it worked

alf45tar, 2023-02-25:
> [www.amazon.com/LILYGO-T-Display-S3-ESP32-S3-Development-Soldered/dp/B0BF542H39/](http://www.amazon.com/LILYGO-T-Display-S3-ESP32-S3-Development-Soldered/dp/B0BF542H39/)
> Just came across this post [...]

When BPI Leaf ESP32-S3 board will works I will support also the T-Display S3.

alf45tar, 2023-02-25:
WiFi fixed. It works pretty well.

rigr, 2023-02-26:
yes, It's nice - comes even with some more storage than the previous esp32 development board and some more pins. But still needs to have USB-MIDI working.

rigr, 2023-03-12:
Just a little warning - the cables I got with the TTGO-T-display can be plugged into the connectors of the leaf BUT the poles are reversed - + is connected to - and - is connected to +. Fortunatelxy the leaf does not emmit electrical smoke but just ignores a bettery connected the wrong way.

Jelle7and9, 2023-07-11:
Is USB MIDI confirmed working on the BPI LEAF ESP32-S3 nowadays? Or should I add a Pro Micro for that?

alf45tar, 2023-07-12:
Yes, it is. It has been released last week with version 3.1.7 for sponsor only. It works for me and I didn't received any feedback at the moment.
It is only USB MIDI device mode. Host mode is not yet implemented.

closed 2023-09-07 by alf45tar