alf45tar/PedalinoMini #423: Can not see pedalinomini on ipad with amplitude 5

jimhiggs, 2023-01-26:
I compiled with the devkit-vi env. Ble is enabled and using Midi+BTLE app on my android phone to scan, I am seeing a device 7512CFA4 when I power up my pedal so seems like it is working.
I run amplitude 5 and Settings - Control - Irig BlueBoard and get "Select the device you want to use from the following list" but It does not find my board.
I am not very good with ipad as I rarely use it but have just installed amplitude 5 and would like to try the BlueBoard configuration (which I have loaded into a profile).
Any ideas?

jimhiggs, 2023-01-27:
I decided not to use Blueboard connect. I installed the korg Bluetooth midi connect and with that running and connected to my pedalboard I can configure all my pedals however I need in amplitude.

alf45tar, 2023-01-27:
PedalinoMini simulates only MIDI over Bluetooth mode 1 (press B on power on) and mode 2 (press C power on).
Bluetooth mode (press A on power on) is a proprietary mode that requires IK app to connect.

Connection between PedalinoMini and Windows/macOS/iOS/Android is standand BLE-MIDI protocol that requires standard connection method depending by OS in use. Connect PedalinoMini before switch to Amplitube.

alf45tar, 2023-01-27:
To connect an iOS device there are a lot of tools. Here below just a couple.

midiWrench see here around 2:00
https://www.youtube.com/watch?v=U14sLy3NBms

Garageband see here around 0:45
https://www.youtube.com/watch?v=pCNSvJ9QiDs&t=100s

jimhiggs, 2023-01-27:
Thanks for the update. I am using your bluebird config profile as a starter and it has given me some good ideas but no point in using amplitude in blueboard mode as std midi is more flexible it seems.

closed 2023-01-27 by jimhiggs