alf45tar/PedalinoMini #426: When Change bank Leds set in global are turned off

jimhiggs, 2023-01-28:
Version 3.0.9 (do you have a separate issue queue for Development versions?)
I am using most switches in bank 0 so that they are available regardless of bank setting,  which works as I expected. 
But Leds I have lit  are turned off when the bank is changed. I expected all leds to remain in the state I set them to.
ie 
I use a switch labelled Reverb (Using Sequence) to send the Reverb activate CC, light the led next to the switch and then switch bank to the bank reserved for reverb to give access to the modifying jogs etc. 
If I then press the Switch for Delay the Reverb light is turned off as the bank is changed to Delay etc.

jimhiggs, 2023-01-28:
Rather than clutter the issues with dev problems I will add them to this issue. Let me know if that is wrong.
The config editor does not know about Step by Step and this seems to cause problems when I try saving a config.

alf45tar, 2023-01-29:
This is the only place to report issue for both versions. Discussion is always public.
Did you updated the system image?
Config editor works on configuration file. It cannot edit the current configuration at least for the moment.

jimhiggs, 2023-01-29:
Yes I have updated the system image

mstp01, 2023-04-06:
I have the same issue - LED associated with latched MIDI action is turned off when switching a bank (but MIDI is still working as needed). Anyone found some workaround on that matter?

Secondly, is there any way to define LED color at reboot? In my case, Bank 1 is starting (which is fine) but its LED has a wrong color (I've set up this LED as indicator of current bank - green for bank 1, red for bank 2 etc.)