alf45tar/PedalinoMini #427: Release action judgment problem

bobvc133, 2023-02-04:
I found that when using single button and double button actions at the same time, there will be release action judgment problems,
For example, button1, button2, and button1+button2 all use the Press&Release action. When the button1 and button2 buttons are pressed at the same time, button1+button2 Press will be executed, but if the button1 and button2 buttons are not released at the same time, the Release of button1 and button2 will be executed at the same time.A Relaesa judgment needs to be added here, the Release action should correspond to the Press action, although it may be a misoperation.A Relaesa judgment needs to be added here, the Release action should correspond to the Press action, although it may be a misoperation.

https://user-images.githubusercontent.com/91867813/216777068-1bc9e00f-b417-427d-92cb-504ea0692554.mp4

alf45tar, 2023-02-06:
It works as designed. Two buttons are considered simultaneous pressed if the second is pressed within a certain gap from the first (50ms by default). If it not pressed within the gap is considered as single pressed or released.

What do you mean with "the Release action should correspond to the Press action, although it may be a misoperation"?

bobvc133, 2023-02-07:
> It works as designed. Two buttons are considered simultaneous pressed if the second is pressed within a certain gap from the first (50ms by default). If it not pressed within the gap is considered [...]

I press the AB button at the same time to perform the Press&Release action, but if the release is not kept within the gap (50ms by default), the single release action of the A and B buttons will be executed. At this time, the Press&Release action of the AB button will only take effect. Press, becomes For the latch effect, both the A and B buttons perform a separate Release action, which also becomes a latch effect. Conversely, when the AB button is pressed at the same time without keeping it within the gap (50ms by default), the single-press action of the A and B buttons will be executed, but when the AB button is released at the same time, the Realeae action of the AB button will be executed. The probability of this kind of misoperation is high, and the momentary effect of Press&Release also becomes a latching effect. Although it works as designed, I may need more misoperations to correct it, and it will send multiple error messages at the same time. What I mean is that when using the Press&Release action, if I press the AB button at the same time, the Release of the AB button should be executed regardless of whether it is kept in the gap or not. If I don't press the A and B buttons at the same time, even if I release them at the same time, it should perform a single release action of the A and B buttons, so that I can easily correct it even if it is a mistake.

alf45tar, 2023-02-07:
Unlikely the combinations with 2 buttons are a lot. For example, should the actions on release be executed on A or B? Someone can say A but someone else can say B. Someone can say on the first button released but someone else can say on the second button released. There is no right or wrong answer.
I think we are going far from the scope of the project.
If there is a risk of misoperation reduce or avoid the risk. You can increase the gap interval or trigger event on press only.

bobvc133, 2023-02-07:
It may be a problem of translation, what I expressed is not the same problem as your explanation.

alf45tar, 2023-02-07:
> I press the AB button at the same time to perform the Press&Release action, but if the release is not kept within the gap (50ms by default), the single release action of the A and B buttons will be executed. 
 > At this time, the Press&Release action of the AB button will only take Press, exhibits a latch effect, both the A and B buttons perform a separate Release action, which also becomes a latch effect. 

Press&Release actions is exactly the same of one Press action + one Release action. Press&Release do not add any new feature. It is a just to be more compact.
The latch effect is when 2 consecutive physical press of the button send 2 different messages (the first on Press and the second on Release like a latch switch).
In your case there is no latch effect because both pressure send the same messages.

> What I mean is that when using the Press&Release action, if I press the AB button at the same time, the Release of the AB button should be executed regardless of whether it is kept in the gap or not.

If the the release is not in the gap when the Relase action should be triggered? On A release, on B release, on the first release (A or B whichever come first) or on the second release (A or B whichever come later)? 

>If I don't press the A and B buttons at the same time, even if I release them at the same time, it should perform a single release action of the A and B buttons, so that I can easily correct it even if it is a mistake

You are saying that current event can trigger a different action depending of the previous event. And what about if other button events are mixing? Again there are too much combinations.

bobvc133, 2023-02-08:
First of all, I am very clear about the principles of instant effects and latch effects. Secondly, I have no doubts about when to trigger the Release operation. Third, event mixing is not my problem. 

Let me give an example to illustrate the problem I want to express. Control 1 (button A), control 2 (button B), and control 3 (button A+B) all use the Press&Release event. 

1. Press control 3 (press button A+B at the same time) to execute the press event of control 3, but when it is released ( Release the button A+B beyond the gap, which is a misoperation), and the Release event of control 1 and control 2 is executed, and control 3 will maintain the Press effect at this time. 

2.  Pressing control 3 (button A+B is pressed beyond the gap, which is a misoperation) executes the press event of control 1 and control 2, but when it is released (button A+B is released at the same time), the execution is The Release event of control 3, at this time control 1 and control 2 maintain the Press effect. 

The above two misoperations cause the Press&Release event of the three controls to become a separate Press event or Release event. My suggestion is to ensure the integrity of the Press&Release event of each control, even if it is mishandled. 

1. If you press control 3 (press buttons A+B at the same time) to execute the press event of control 3, then even if the buttons A and B are released beyond the gap when you release it, the Release event of control 3 should be executed. 

2. Press control 3 (button A+B is pressed beyond the gap, which is a misoperation) to execute the press event of control 1 and control 2, so when you release it, even if you release buttons A+B at the same time, it should also be executed Release events of Control1 and Control2. In this way, even a misoperation will not cause Press&Release to be split and executed.

[image]
[image]

https://user-images.githubusercontent.com/91867813/217562805-50f5bef5-b711-4b90-8739-668973cb8cd1.mp4

https://user-images.githubusercontent.com/91867813/217562932-aebf6b57-f59a-4405-8f92-a3a504c5177e.mp4

closed 2024-05-16 by alf45tar