alf45tar/PedalinoMini #428: Build Error | 2.4″ OLED Display | Configuration Files

TarFilarek, 2023-02-10:
Hello!

- **_1st question_**

I'm trying to upload the software to the newest version (3.0.9), and I constantly get these red errors during building process.

`src/Controller.h:1271:40: error: no match for 'operator[]' (operand types are 'CRGB [17]' and 'CRGB')`

`src/Controller.h:1271:98: error: no match for 'operator[]' (operand types are 'CRGB [18]' and 'CRGB')`

`*** [.pio\build\esp32dev\src\PedalinoMini.cpp.o] Error 1`

Whole terminal message:

[Pedalino.txt](https://github.com/alf45tar/PedalinoMini/files/10709404/Pedalino.txt)

What should I do?

- **_2nd question_**

I've changed old OLED Display from [1.3″](https://pl.aliexpress.com/item/32683739839.html?spm=a2g0o.order_list.order_list_main.10.f6d91c24VWGOjH&gatewayAdapt=glo2pol) to [2.42"](https://pl.aliexpress.com/item/1005003950796751.html?spm=a2g0o.order_list.order_list_main.10.704f1c24JHXe5r&gatewayAdapt=glo2pol), and everything worked immediately, but with a little problem.

As You can see below, all displayed pixels are moved to the right.

[image]

How can I repair it?

- **_3rd question_**

Some time ago I've created a large Configuration File, with lots of actions. Pedalino worked with it without a problem. I saved it in two separate parts - one for actions only and second for all the other settings. Now I'm unable to upload it back (the one with actions), because the Pedalino resets itself each time. The file was too large, so I had to delete 30-40 actions from it.

Is there any other way, to upload larger Configuration Files to the Pedalino?

alf45tar, 2023-02-11:
Question 2

Maybe it has a SH1106 controller instead of the SSD1306.
Edit your env in platformio.ini, comment the first line and uncomment the second

```
	-D SSD1306WIRE
;	-D SSH1106WIRE 
```

alf45tar, 2023-02-11:
Question 3

You can split large configuration file into small one and upload in sequence.

alf45tar, 2023-02-11:
Question 1

Fixed in version 3.0.10 just released

TarFilarek, 2023-02-11:
> Maybe it has a SH1106 controller instead of the SSD1306.

It was actually the other way around, but yes, You were right - that was a cause of it. Now everything works great!

> Fixed in version 3.0.10 just released

Tested and can confirm - works like a charm!

> You can split large configuration file into small one and upload in sequence.

I will definitely give it a try once I get used to the new "Controls" system.

Thank You so much for Your help!

At first I was a little bit puzzled about "Control" cell, because it was already defined in "group title" just over it, but I gave it a try and found out, that You can use it to change the main Control number. I guess that was its purpose and it is a great feature - I can use it to repair my old Configuration Files created before this modification.

I would propose only to consider renaming this cell from "Control" to something like "Change Control to...". It would make it a little bit more obvious.

I made a little test, and found out, that everything change places nicely except for the "Slot" number. It stays where it was before.

(Before Saving -> After Saving)

[image]

The "Slot" option probably doesn't work right now (or maybe I don't know how to use it), but it's just a little detail to keep in mind for the future upgrades. 

The other little thing I found out, is that the "Control Change" Send Type doesn't allow us to modify the left side values (Value, Tag and Off LED Colour). Cells are not visibly greyed out, but You can see it above.

And lastly - some enhancement ideas:

- introducing "groups" in place of the current "Control" selection window. 

[image]

If we could create our own groups (or add new ones to the current window) like "bottom row" and "top row" consisting of chosen control numbers, it would make it easier to "visualize" and understand them while modifying. Previous "Pedals" were working in a kinda similar way, and it was really helpful.

- introducing "saved colours" to speed up the process of setting up new Controls with the same colour palette.

[image]

TarFilarek, 2023-02-12:
> You can split large configuration file into small one and upload in sequence.

I just gave it a test, and unfortunately it doesn't work. 

Uploading new Configuration File (Actions in Bank 2 only) to the existing one (Actions in Bank 20 only) cleared everything from Bank 20 and placed new actions into Bank 2.

I don't know if it is possible to implement, but it would be a really great feature. 

It could be operated by additional button with something like "Add to existing" in addition to the current "Apply" one.

I was playing with Banks for a while, and I've noticed few things. Maybe You will find some of them worth a thought.

Let say I want to exchange the order of Bank 19 and Bank 20. In order to do that, I need to:

1. Duplicate Bank 20 to a different, empty one (e.g. Bank 2).
2. Manually clear the whole Bank 20.
3. Duplicate Bank 19 to Bank 20.
4. Change name of the Bank 20.
5. Manually clear the whole Bank 19.
6. Duplicate Bank 2 (Old Bank 20) to Bank 19.
7. Change name of the Bank 19.
8. Manually clear the whole Bank 2.

And here are a few things, that would make it easier and quicker:

- Option to clean a whole bank. 

E.g. - "Clear Bank" button next to the "Duplicate Bank To" button.

- Option to change the Bank number.

E.g. - "Bank Number" cell next to the "Bank Name". It could work exactly as current "Control" cell in Actions.

- Option to move whole bank to a different Bank (basically a same thing as above).

E.g. - "Move Bank To" button next to the "Duplicate Bank To" button.

- Operations on selected Actions (instead of a whole bank).

E.g. - Small check boxes next to all of the actions and "Move Selected To", "Copy Selected To" and "Delete Selected" blue buttons next to the "Duplicate Bank To" button.

Additionally:

- "Duplicate Bank" doesn't duplicate banks name.

alf45tar, 2023-02-13:
Did you try to drag&drop the bank name/number? You can reorder the banks.

TarFilarek, 2023-02-13:
> Did you try to drag&drop the bank name/number? You can reorder the banks.

Damn, i totally forgot about it. You are right - it works. 

At first i thought it didn't, cause after saving the bank was going back to the previous spot, but it works, if I click the relocated Bank after moving it. Then, after saving, it stays in a new spot. Thank You!

Sorry for bothering You with all of these, but last thing - what about those greyed out/blocked "Press | Control Change" left side parameters (Value/Tag/Off)? Is it a bug?

[image]

alf45tar, 2023-02-13:
It isn't a bug. I changed the design to be more clear.
If you want to enable the left side select Release event.
If you want to have both select the Press&Release event.
If you want to simulate a latch button with a momentary switch use a sequence and Step by Step+ action.

alf45tar, 2023-02-13:
I am evaluating to rename "Control" to "Change Control To" because it could be confused with MIDI Control Change.

TarFilarek, 2023-02-14:
> If you want to simulate a latch button with a momentary switch use a sequence and Step by Step+ action.

Frankly, I belive the previous design was much, much better. I think, right now, it is actually really discouraging to create something more complex.

Let's say I would like to use 6 buttons to take me to 6 different banks (pedalboards), each one consisting of 12 different ON/OFF actions. I had something similar in the past.

It would take 72 different sequences with a total of 144 steps - each one manually created, cause there is no "Copy" or "Duplicate". I can't even use a color picker, cause the desired colour is in a previous sequence.

Previously I would have to simply create one full bank manually (12 actions), and then I would duplicate it 5 times, change the CC values and it would be done. Easily and quickly.

Right now, when I open a random sequence, I simply have no idea what it does in the bank, what is it's tag, button type, nothing. And same in the bank - I see only a sequence number. I can't see if the value is what I would like it to be, or if a LED Color is set properly. It would take hundreds of additional clicks, "Saves", copy-and-pastes, notes and double/triple checks to make it all work as before.

If we look at a real-world pedalboard, its main function is to simply turn effects ON and OFF with a press of a switch. Making it so time-consuming and difficult to replicate is a real downgrade from the previous version. Please - consider bringing it back. It was really awesome.

alf45tar, 2023-02-14:
Did you considered to use latch switches instead of momentary?

alf45tar, 2023-02-14:
I am evaluating to simulate the latch effect on pedal level and not on actions.
I can add a flag in Pedals configuration for Momentary switches called "Simulate Latch".
The effect of the flag is:

1. all the Release events are filtered
2. first press (and next odd) trigger a Press event
3. second press (and next even) trigger a Release Event
4. double press and long press are disabled

TarFilarek, 2023-02-14:
If I were building it right now - sure, I could, but is has been finished in [2021](https://github.com/alf45tar/PedalinoMini/issues/284) in accordance to the older versions.

> I can add a flag in Pedals configuration for Momentary switches called "Simulate Latch".

It would definitely be a great addition, worth introducing, but in my config. I'm using multiple different actions on each one of the buttons (3 ladders of 6) and that includes double clicks, long presses and releases so... for the time being, I will just move back to the 2.5.4 firmware to use my old config. files and I will try to make something out of 3.x.x. in the future. 

Thank You!

alf45tar, 2023-02-14:
Version 3.1.0 with latch emulation for momentary switches is ready if you want to give a try.
Double press and long press are supported.

Reset to factory default because EEPROM changed.

TarFilarek, 2023-02-22:
Hello

> Version 3.1.0 with latch emulation for momentary switches is ready if you want to give a try.

Thank You!

I've just tried to check it (v3.1.1) and I've got this error (during building phase).

> In file included from .pio/libdeps/esp32dev/TFT_eSPI/TFT_eSPI.cpp:24:0:
> .pio/libdeps/esp32dev/TFT_eSPI/Processors/TFT_eSPI_ESP32.c: In member function 'bool TFT_eSPI::initDMA(bool)':
> .pio/libdeps/esp32dev/TFT_eSPI/Processors/TFT_eSPI_ESP32.c:803:3: error: 'spi_bus_config_t' has no non-static data member named 'data4_io_num'
>    };
>    ^
> *** [.pio\build\esp32dev\lib688\TFT_eSPI\TFT_eSPI.cpp.o] Error 1

Third and the last line are red.

alf45tar, 2023-02-24:
https://github.com/Bodmer/TFT_eSPI/discussions/2437

alf45tar, 2023-02-24:
Fixed using version TFT_eSPI@2.5.21

bobvc133, 2023-02-27:
> Version 3.1.0 with latch emulation for momentary switches is ready if you want to give a try. Double press and long press are supported.
> 
> Reset to factory default because EEPROM changed.

Use the voltage ladder mode, after the Simulate latch function is turned on, all buttons will become a latch effect

closed 2023-04-10 by alf45tar