alf45tar/PedalinoMini #432: 2.5.4 Missing features

Ratterbass, 2023-02-26:
I have tried installing the sponsor version firmware both via platformIo and via webinstaller+manual update, but there seem to be some features missing. 

1) I can't find send control snap under actions
2) I can't find simultaneous button press under triggers

Is there something missing in the installation files, do I have a bug? User error?

Thanks in advance!

[image]
[image]
[image]
[image]

alf45tar, 2023-02-26:
2.5.4 is the public version. The latest sponsor version is 3.1.1 available here https://github.com/pedalino-sponsors/PedalinoMini.

Ratterbass, 2023-02-27:
> 2.5.4 is the public version. The latest sponsor version is 3.1.1 available here https://github.com/pedalino-sponsors/PedalinoMini.

I don't see anything more recent than 2.5.4 in that repo. 
[image]

When I manually build 3.1.1 from the src files and upload it to the ESP, the chip is stuck in a bootloop:

[image]

The manually built firmware.bin file is just over 2M, so manual update through the webUI doesn't work either. 

Any ideas?

alf45tar, 2023-02-27:
Did you follow the instructions https://github.com/alf45tar/PedalinoMini#how-to-build-and-upload?
Just replace the repository to clone.

Ratterbass, 2023-02-27:
Yes, I did. I deleted the whole thing and tried both manual download as well as git:clone. 

Always the same result: boot loop

alf45tar, 2023-02-27:
Did you reset EEPROM to factory default?

Ratterbass, 2023-02-27:
I redid the whole procedure, including flash erase on a windows PC instead of a Mac and it went smoothly and the controller is now running. 

Seems like MacOS (13.2.1) needs some special attention when flashing?

alf45tar, 2023-02-28:
Weird because I am on MacOS 13.2.1 too.

alf45tar, 2023-03-01:
In the next version I will delete ports from platform.ini because auto detection works fine.

closed 2023-03-01 by alf45tar