alf45tar/PedalinoMini #433: Restart the device after pressing the buttons

e36910, 2023-03-04:
After switching presses I see the same problem, the device restarts, sometimes it works fine, and sometimes by pressing a button the device restarts, I thought that the problem was with the board, but when I replaced it, the problem remained the same. Buttons are soldered through the resistors 10k in series, no problems with the power supply, tried through the usb and through the pin 5v solder. What can be the problem?

alf45tar, 2023-03-04:
Can you post here the log?
The pull up resistor is not is not in serie. However is should not be the reason of restart.

e36910, 2023-03-04:
My example - https://youtu.be/Fg-qYdnU9uI

e36910, 2023-03-04:
how to create log?

alf45tar, 2023-03-04:
The debug information are printed automatically on USB UART. You can use Visual Studio Code and Monitor option or any other program can read the serial port via USB.

e36910, 2023-03-04:
I don't know how to do it

e36910, 2023-03-04:
PROGRAM CHANGE.....Program   0.....Channel  1
Pedal: 1     Button: 2    EventType: 1     ButtonState: 1
Pedal: 2     Button: 2    EventType: 0     ButtonState: 0
Action: 1
CONTROL CHANGE.....Code   1......Value   0.....Channel  1
LED COLOR.....Led  0......RGB Color #050005
Pedal: 2     Button: 2    EventType: 1     ButtonState: 1
Pedal: 2     Button: 3    EventType: 0     ButtonState: 0
Action: 2
CONTROL CHANGE.....Code   2......Value   0.....Channel  1
LED COLOR.....Led  0......RGB Color #000505
Pedal: 2     Button: 3    EventType: 1     ButtonState: 1
Pedal: 2     Button: 3    EventType: 0     ButtonState: 0
Action: 2
CONTROL CHANGE.....Code   2......Value 127.....Channel  1
LED COLOR.....Led  0......RGB Color #000505
Pedal: 2     Button: 3    EventType: 1     ButtonState: 1
Pedal: 2     Button: 3    EventType: 2     ButtonState: 1
Pedal: 2     Button: 4    EventType: 0     ButtonState: 0
Action: 3
CONTROL CHANGE.....Code   3......Value 127.....Channel  1
LED COLOR.....Led  0......RGB Color #050400
Pedal: 2     Button: 4    EventType: 1     ButtonState: 1
Pedal: 1     Button: 2    EventType: 0     ButtonState: 0
Action: A
PROGRAM CHANGE.....Program   0.....Channel  1
Pedal: 1     Button: 2    EventType: 1     ButtonState: 1
Pedal: 1     Button: 3    EventType: 0     ButtonState: 0
Action: B
PROGRAM CHANGE.....Program   1.....Channel  1
Pedal: 1     Button: 3    EventType: 1     ButtonState: 1
Pedal: 1     Button: 3    EventType: 2     ButtonState: 1
Pedal: 1     Button: 4    EventType: 0     ButtonState: 0
Action: C
PROGRAM CHANGE.....Program   2.....Channel  1
Pedal: 1     Button: 4    EventType: 1     ButtonState: 1
Pedal: 1     Button: 3    EventType: 0     ButtonState: 0
Action: B
PROGRAM CHANGE.....Program   1.....Channel  1
Pedal: 1     Button: 3    EventType: 1     ButtonState: 1
Pedal: 1     Button: 3    EventType: 2     ButtonState: 1
Pedal: 2     Button: 3    EventType: 0     ButtonState: 0
Action: 2
CONTROL CHANGE.....Code   2......Value   0.....Channel  1
LED COLOR.....Led  0......RGB Color #000505
Pedal: 2     Button: 3    EventType: 1     ButtonState: 1
Pedal: 2     Button: 3    EventType: 2     ButtonState: 1
Pedal: 1     Button: 2    EventType: 0     ButtonState: 0
Action: A
PROGRAM CHANGE.....Program   0.....Channel  1
Pedal: 1     Button: 2    EventType: 1     ButtonState: 1
Pedal: 1     Button: 3    EventType: 0     ButtonState: 0
Action: B
PROGRAM CHANGE.....Program   1.....Channel  1
Pedal: 1     Button: 3    EventType: 1     ButtonState: 1
Pedal: 2     Button: 4    EventType: 0     ButtonState: 0
Action: 3
CONTROL CHANGE.....Code   3......Value   0.....Channel  1
LED COLOR.....Led  0......RGB Color #050400
Pedal: 2     Button: 4    EventType: 1     ButtonState: 1
Pedal: 1     Button: 4    EventType: 0     ButtonState: 0
Action: C
PROGRAM CHANGE.....Program   2.....Channel  1
Pedal: 1     Button: 4    EventType: 1     ButtonState: 1
Pedal: 1     Button: 2    EventType: 0     ButtonState: 0
Action: A
PROGRAM CHANGE.....Program   0.....Channel  1
Pedal: 1     Button: 2    EventType: 1     ButtonState: 1
Pedal: 1     Button: 4    EventType: 0     ButtonState: 0
Action: C
[... 188 lines cut ...]
USB MIDI started
DIN MIDI started
BLE MIDI service advertising started
Skipped connection to last AP and/or SmartConfig/WPS setup
SYSTEM_EVENT_WIFI_READY
AP MelatoninPedals started with password 1234567890
AP SSID     : MelatoninPedals
AP PSK      : 1234567890
AP MAC      : 40:22:D8:EB:1D:45
AP IP       : 192.168.4.1
SYSTEM_EVENT_AP_STOP
SYSTEM_EVENT_AP_START
SYSTEM_EVENT_AP_START
Channel     : 1
Connect to MelatoninPedals wireless network with password 1234567890
mDNS responder started
HTTP server started on port 80
Connect to http://Melatonin.local/update for firmware update
Connect to http://Melatonin.local for configuration
ipMIDI started
RTP-MIDI started on port 5004
OSC server started on port 8000
Internal Total Heap 269248, Internal Free Heap 97704
SYSTEM_EVENT_STA_START
SYSTEM_EVENT_SCAN_DONE

e36910, 2023-03-04:
[image]

e36910, 2023-03-04:
[image]

alf45tar, 2023-03-05:
It looks like it is not connected with the software because I do not see any software exception error like "division by zero" or "illegal address".
It is a normal log when someone pressed the reset. I suggest to check your hardware connections, short circuit and power supply.
If you post some shots of internals I can have a look?

e36910, 2023-03-05:
[image]

e36910, 2023-03-05:
[image]

closed 2023-09-07 by alf45tar