alf45tar/PedalinoMini #434: individual tag per control, not just per pedal

Ratterbass, 2023-03-06:
Is there a way to display individual tags per *control* not just per pedal? Example:

if you use 3 switches with momentary3 configuration you only get to see one tag per pedal, so if you have 2 momentary3 pedals configured you only get two tags. 

One workaround would be to assign "empty" controls to the pedal ports that are not in use, but those brings two issues:

1. You still can't reassign tags for pedal 1 and 2
2. Only works if the other ports are not in use

So is there a way to assign tags to every control? Or to force to show tags?

Ratterbass, 2023-03-06:
Ok, so this works by using the "empty" controls from control 1 to 6 as thos ehave priority over others for the tags and won't be overwritten.

alf45tar, 2023-03-17:
The idea behind Slot is to assign the Action to one of the 6 display slots available. Unlikely it is never been developed and the slot are assigned one by pedal.

zavorra, 2023-12-25:
@alf45tar, for my own build (as of now) I am developing the "Slots by Action" instead that "Slot by Pedal" feature

this is a patch (agaist tag 3.2.0) just a proof of concept to show my idea,  as of now I only modified the upper row of slots. Before going on I would like to ask you if this could be a correct approach.

[POC_slot_from_actions.patch.zip](https://github.com/alf45tar/PedalinoMini/files/13765689/POC_slot_from_actions.patch.zip)

If you think this can be also useful to you  I will be happy to follow your direcitions, if any!

Z

zavorra, 2023-12-31:
This is a (sliightly) more complete patch (agaist tag 3.2.0).

It is working fine on my use case (commanding a MOD DWARF via PC or CC or sequences) with a lilygo-t-display-s3 based pedalino, 8 push buttons and 2 analog.

 

[more_slots_by_actions.patch.zip](https://github.com/alf45tar/PedalinoMini/files/13801462/more_slots_by_actions.patch.zip)
(url)

Jelle7and9, 2024-01-24:
Eureka moment here...
I was tinkering and just find out how this works with the slots.
I didn't get it working with a 6 button Ladder and today I was trying to assign Pedal 7 to a slot and 4x Momentary 2 configs. 
Know I understand this can't work that way.
I would like to assign a Slot to an Button Action. So every Button can be assigned to a Slot as you wish.

@alf45tar 
Will you consider inplementing programmable Slot assignment by Button/Action?