alf45tar/PedalinoMini #437: The configuration file cannot be created, exported, applied and saved

bobvc133, 2023-03-11:
When there are many BANK configurations, it can be configured and used normally, but it cannot be saved to the available configuration list, or it will be saved as an empty file, or the device will restart. Use the configuration editor to configure directly, and it can be saved to the available configuration list, but the application is saved to The current configuration is either invalid or causes the device to restart. The firmware version is 2.5.4. Because the configuration is more complicated, the 3.0 firmware has not been tested yet.
[BANK.txt](https://github.com/alf45tar/PedalinoMini/files/10948821/BANK.txt)

alf45tar, 2023-03-17:
**Short answer**

File is too big for the available memory.

**Long answer**

To deserialise your BANK file 68Kb are requested as minumum by ArduinoJSON library. Unlikely not more than 65K are available.

**Solution**

Split the configuration file in multiple short files ad apply them one be one.

bobvc133, 2023-03-20:
> **Short answer**
> 
> File is too big for the available memory.
> 
> **Long answer**
> 
> To deserialise your BANK file 68Kb are requested as minumum by ArduinoJSON library. Unlikely not more than 65K are available.
> 
> **Solution**
> 
> Split the configuration file in multiple short files ad apply them one be one. Your file has "only" 205 Actions.

1. I have configured 16 BANKs. Although the file size does not exceed 65K, the above problems still occur.
2. Split the configuration file into two configuration files, BANK1-10 and BANK11-16. The other configuration file will be overwritten when the application is saved. Please tell me how to do it correctly.

[Ampero-part1.TXT](https://github.com/alf45tar/PedalinoMini/files/11020523/Ampero-part1.TXT)
[Ampero-part2 .TXT](https://github.com/alf45tar/PedalinoMini/files/11020525/Ampero-part2.TXT)

alf45tar, 2023-03-22:
It depends of the dimension of the maximum allocation block in heap.
Check the log for error like this one

Memory allocated for JSON document: 42076 bytes
Reading /Ampero-part1.TXT.cfg from SPIFFS ... deserializeJson() failed with code NoMemory

alf45tar, 2023-03-22:
To not override configuration comment the following line

https://github.com/pedalino-sponsors/PedalinoMini/blob/706d26be2fbb66693f1df827ad8bf5e0f546e8b6/src/Config.h#L618

alf45tar, 2023-07-03:
Configuration file can be appended to current profile (only Actions) in version 3.1.5.

closed 2023-07-03 by alf45tar