alf45tar/PedalinoMini #441: Spurious responses from resistor ladder

mosswind, 2023-03-31:
I did the #7 boot up mode, and pressed each switch, and they all showed up on the line as described in the instructions.  But now I seem to be having events triggered at random, or long delayed presses happen. Is there any way to de-bug this?  Loving the PedalinoMini, BTW!  Can't wait to iron out these couple of bugs.

alf45tar, 2023-03-31:
Which schematic of ladder did you used? Are the value read by calibration "equally" spaced or are they closed to each other?

mosswind, 2023-03-31:
Hi alf45tar-

I used the 10k resistors.  They’re 1% tolerance, 1/2 watt resistors.  I also have some other questions for you, but let’s stick with this for now.

Michael

alf45tar, 2023-04-01:
After ladder calibration which are the value reported in Options for ladder levels?

mosswind, 2023-04-01:
The numbers are 515, 690, 785, 848, 910, 979 and 1023.

Michael

alf45tar, 2023-04-01:
Did you used th following schematic?
<img width="429" alt="Screenshot 2023-04-01 at 21 48 51" src="https://user-images.githubusercontent.com/3542
6671/229311275-66b6c74c-e77e-41fd-8554-3a844ff74b63.png">

Did you connected TP to a analog pin and RING to a digital pin?

mosswind, 2023-04-01:
> Did you used th following schematic?
>  <applewebdata://3A2DB5C3-67CD-4FE2-954F-54644E6D2FE4>
> Did you connected TP to a analog pin and RING to a digital pin?

Hi- I could not make any sense of the number of the schematic you sent, but it is the one at the site now-  dated 2022-07-02.  And yes, I crossed the pins for tip and ring as instructed.  Do you think I should try one of the other voltage ladders, like the 2k, 3k, 5.1k, 10k and 30 k? 

If you would consider doing a phone consultation, perhaps we could get to the bottom of it.  You could talk me through what ever test I could do with a multimeter.  I would be happy to compensate you for getting this working.  I have a number of projects that I am trying to do with the PedalinoMini, and this would free them all up for success.

My website is soundscapeboston.com <http://soundscapeboston.com/>, so you can see a little bit about me.  I also do sound installation art- here is a link to a video of one of my pieces: https://www.youtube.com/watch?v=Ix8CDOWI_TY <https://www.youtube.com/watch?v=Ix8CDOWI_TY> 

Best,

Michael

alf45tar, 2023-04-02:
Hi Michael,
phone consultation is ok for me. Is Google Meet ok for you? I do not accept any compensation. I accept only donation via GitHub Sponsor that you already did. Considering the different time zone the best schedule for me is from 1 PM to 5 PM your time.

Before connect can you check that analog inputs must have a pull-up resistor as per schematic or they are disabled by software in Pedals page? This is the main reason of random events trigger.

Kind regards
alf45tar

mosswind, 2023-04-02:
Hi alf45tar-

That is probably the answer!  I saw that on the schematic, but somehow, it did not register- I thought that everything was on the  Minelli board somehow.  This is where my lack of thorough electronics knowledge can be my downfall.  I used to be a ham operator when I was a kid, but that was many years ago.  Otherwise, I’m pretty much winging it on what I learned then.  I dd try  another pattern of resistor ladder last night (2k, 3k etc), and had the same problem.  

Are weekdays good for you to do the Google meet consult, if that does not straighten it out?

Best,

Michael

alf45tar, 2023-04-02:
Weekdays are ok for me.

mosswind, 2023-04-03:
How would tomorrow at 3:30 eastern be for you?  Status as of this moment- Can’t get the PedalinoMini to go through the boot mode for the resistor ladder with the pull-up voltage in place.

Michael

closed 2023-04-10 by alf45tar

mosswind, 2023-04-21:
Alf-

Excellent assistance.  All is working very well.  A million thanks!

Michael

rigr, 2023-05-03:
- great! Would anyone of you mind to share what was the problem and how it was sorted?
- Anyways, thanks to alf45tar!

mosswind, 2023-11-18:
Hi alf45tar-

Michael here, the sax player/sound installation artist that you communicated with before.  I am back being active after having moved, etc., and I have a question.  I am in a new home with of course a new WiFi set- up.  I need to change the address for the Pedalino mini I was working with last time.  I would rather not erase the whole board- as you may recall, you set me something special up to give me a bunch of note ons.  Can we work together to get me back to that point?  I will be happy to donate to the project on Github- a little extra than my subscription.

Best,

Michael