alf45tar/PedalinoMini #443: T-Eight firmware update issue

bgen5pax, 2023-05-01:
Just noticed that newest 3.1.5 upgrade does not complete normally via web installer on T-Eight, 2.5.2 installs just fine. Only way to upgrade is from PC upload through Options.
Again, thanks for all your efforts alf45tar.

alf45tar, 2023-05-01:
I know. There is something that it is changed. Looking into it.
Use PlatformIO to update the firmware.

alf45tar, 2023-05-01:
Fixed. You can give a try.

bgen5pax, 2023-05-01:
Seems to work OK, thank you. Minor issue: power off in settings only temporarily shuts it down and after 4-5 sec it's powered on again, no such problem with previous firmware.

alf45tar, 2023-05-01:
Thanks for let me know the bug.
It should be fixed in public version 3.1.6 just released.

bgen5pax, 2023-05-06:
Problem solved, you may close the issue. I also noticed that PSRAM of T-Eight is not used/implemented at all. Is this on purpose?
Again, many thanks for all your efforts.

closed 2023-09-07 by alf45tar