alf45tar/PedalinoMini #449: Missing info on Schematic_PedalinoMini_Sheet-3.svg

Wapata, 2023-06-07:
Hello !
My PedalinoMini bpi-leaf-s3 is on my wifi ! It seem it's a success so far.
But can you please add two words into the https://github.com/alf45tar/PedalinoMini/blob/master/images/Schematic_PedalinoMini_Sheet-3.svg shematic ?
I'm looking for "LADDER" and "LADDER 2" because of this paragraph : https://github.com/alf45tar/PedalinoMini#controls

Thanks a lot ! And bravo because it wasn't so hard to put PedalinoMini into the board using -nearly- only your expalinations 

(then I have to find what the heck are those connectors for SAD/SCL and batterie)

alf45tar, 2023-06-09:
LADDER 2 is an error. It doesn't exist. It should be LATCH 2.

Wapata, 2023-06-09:
Oh. Nice then, thanks for thé answer :-)

closed 2023-06-10 by Wapata