alf45tar/PedalinoMini #451: Can't update from Webui

Keeze, 2023-06-12:
Hi,

I try to update to the newest firmware via the web-interface. After pressin the button update and confirming I get the Progress=page, but nothing happens, even If i wait for a long time.

How can I solve this?

Cheers,
kees

alf45tar, 2023-06-12:
Update firmware from web is not supported in all cases. It depends by board memory and how big is the firmware. The firmware size must be less than 50% of the memory amount.
Which firmware and which board are you using?

alf45tar, 2023-06-12:
Only firmware compiled with

`board_build.partitions = src/partitions_ota.csv`

support web OTA update.

Keeze, 2023-06-12:
[I use the firmware esp32doit-devkit on a ESP32DOIT Devkit ](https://github.com/alf45tar/PedalinoMini/tree/master/firmware/esp32doit-devkit-v1)

alf45tar, 2023-06-13:
Did you try to reset the EEPROM to factory default?

alf45tar, 2023-06-13:
Before any update save configuration, reset to factory default and reload configuration. EEPROM can change without any further advice.

Keeze, 2023-06-13:
Nope, I am not, but looking at the article, on 3.3V I need 1.8K resistors?

Keeze, 2023-06-13:
Resistors have been bodged and testing have been done ;) It seems I have a working pedalboard again. Tnx for your help!