alf45tar/PedalinoMini #453: BLE MIDI with NUX Mighty Air

palonsorodriguez, 2023-07-14:
Hi,
I'm trying to connect my NUX Mighty amp vía Bluetooth to the Pedalino Mini but I cannot find how to.

I'm able to connect to it and able to switch channels using MIDI+BTLE and Midi Controller apps, in addition to the companion App by NUX. For that I have to select the NUX amp as the server, so I understand that my phone acts as the client.

Is it possible to use the Pedalino as BT client? Otherwise how can you connect to it using an amp?

Thanks a lot

Best regards
Pablo

alf45tar, 2023-07-14:
You have to recompile the firmware with `-D BLECLIENT`

palonsorodriguez, 2023-07-19:
Hi Alf45tar,

Thanks very much for your response.
And after recompiling will be there an option in the web UI to connect to the device?

Thanks
Pablo

palonsorodriguez, 2023-07-24:
Great to know.

Thanks again for your response

Pablo

closed 2023-11-22 by alf45tar

palonsorodriguez, 2024-01-28:
Hi Alf45tar,

I have a lilygo T-Display v1.1
I'm struggling to flash the device with the suggestions you mentioned. I tried a number of times
1. When connecting to the computer it doesn't recognize the USB anymore. it errors with "Unknown USB Device - (device descriptor request failed)" . I tried to uninstall the drivers, install them... but I don't have any option. I tried with another computer, but still some problem
2. When trying to upload via OTA (as it's installed Pedalinomini 3.1.6) it errors telling that it doesn't accept firmware.bin files bigger than 2M (and the T-Display is 4MB)

Do you think any option I could use to reflash it?
Would it be possible if no options that you allow for a version that could install over OTA (or if not, let me know how I could do 
it) so I can install the modified firmware?

Note: I tried flashing pedalinomini to a WROOM-32 but is very unstable as it resets every some seconds

Thanks a lot

alf45tar, 2024-01-28:
1. If the computer do not recognize USB anymore there is no solution to update firmware. Using Visual Studio Code requires an USB connection too.
2. OTA works only if the firmware size is less than half of the memory size because it works with 2 partitions. One for the current firmware and one with the new firmware.

Regarding DOIT ESP32 DevKit V1 I can suggest to erase the flash, install the firmware and install system image. Do not forget any of these 3 steps.