alf45tar/PedalinoMini #455: Controls appears from nowhere in 3.1.6

raidolo, 2023-07-27:
Hi,

a lot of times, when I'm editing the controls, when I hit save and then apply a lot of empty controls appears from nowhere, It's like that since a long time ago, I thought it was fixed but it continues to happen, if I don't hit save and reboot the device they are going away.  Additionally the Press and  the Release tag when you copy a bank to another do not work on the copied controls, I have to delete each control in the new bank and then start over. When you have to deal with a lot of baks or pedal it's becoming really frustrating. Pedalino 3.1.6.

Could you help?

Check this:

[image]

alf45tar, 2023-09-07:
I cannot replicate the issue. Did you tried to reset to factory default?

raidolo, 2023-09-07:
> I cannot replicate the issue. Did you tried to reset to factory default?

I believe I did it, because I've flashed with the latest version. I could try once again also because I'm having issues with the configuration saves, it seems that it doesn't save the profiles properly, and I don't know why. Maybe there is something wrong in the eeprom from the previous version?
I'll let you know, thanks for the reply.

bobvc133, 2023-09-09:
> I cannot replicate the issue. Did you tried to reset to factory default?

I've encountered this problem a few times, and it seems to occur when editing other pages and then returning to the action page.

raidolo, 2023-09-15:
> > I cannot replicate the issue. Did you tried to reset to factory default?
> 
> I've encountered this problem a few times, and it seems to occur when editing other pages and then returning to the action page.

Yeah, I was not able to replicate it everytime, so I couldn't write more details about it....

bobvc133, 2023-09-23:
> I cannot replicate the issue. Did you tried to reset to factory default?

This problem has occurred very frequently recently. Controls 1 often has a few more blank events, it seems that there are 6 every time. In addition, events in the subsequent controls often disappear for no reason.

closed 2026-02-25 by alf45tar