alf45tar/PedalinoMini #456: DIN Legacy MIDI workaround

aFunkyBass, 2023-08-19:
I'm writing this post to help anyone who could encounter problems getting the Legacy MIDI to work.

I've built my second Pedalinomini controller and I had some problems getting the MIDI DIN to work powering up the system with a 5V external adaptor. Powering the ESP32 via the USB cable Legacy MIDI was working. That puzzled me a lot!

After monitoring all the levels signal on the MIDI pins with an hoscilloscope, I've realized that the probem was the difference in voltage on the pins. On the schematics the connections are like so:
[image]

Now, the previous schematics is taken from the original MIDI specs which was based on 5v logic devices. Being the ESP a 3.3v unit, you have a signal of about 3.3v on pin 5 and a signal of around 5v on pin 4 (not considering the drop down introduced by the resitors). This difference is probably enough to steadily turn on the opto-isolator LED on the receiver, making the communication void. This didn't happen powering it up via the USB plug, because of the voltage drop down introduced by the ESP internal regulator, in fact I had a signal on pin 4 of about 4.5v.

I solved the problem converting the MIDI out to 3.3v logic using the following schematics taken from [www.midi.org](https://www.midi.org/specifications-old/item/midi-din-electrical-specification)
[image] 

Use the table to get the correct resistors value and feed pin 4 with 3.3v, and you should be good to go. :)

prostealth, 2024-10-17:
does this fix connection on kemper?
mine are working on line 6, roland, etc. but not on kemper. i have behringer v-amp 2 that also not detecting input from pedalinomini.

aFunkyBass, 2024-10-17:
It probably will... Converting your circuit shouldn't be very hard, as the only thing to do is replace the two resistors as follows: 

R4 from 220 to 33 Ohm 
R5 from 220 to 10 Ohm

Also, make sure to feed the R4 resistor connected to PIN 4 with 3.3v instead of 5v. You can get a 3.3v from the ESP board.

alf45tar, 2024-10-17:
Only a test can confirm it because 4.5V is acceptable by MIDI specification (5V+-10%) but not all devices works.

prostealth, 2024-10-18:
just want to confirm..
it work on my behringer V-Amp 2..
thanks man...

alf45tar, 2024-10-18:
Which version is working? 3.3V or 5V?

prostealth, 2024-10-18:
> Which version is working? 3.3V or 5V?

3.3V

aFunkyBass, 2024-10-18:
Cool I'm happy I could help you. :)

closed 2026-11-02 by alf45tar