alf45tar/PedalinoMini #457: Error building in VSCode

DWSQUIRES, 2023-08-24:
Hi there,

I'm trying to install 3.1.7 and I get this error

In file included from .pio/libdeps/esp32dev/OSC/SLIPEncodedBluetoothSerial.cpp:1:0:
.pio/libdeps/esp32dev/OSC/SLIPEncodedBluetoothSerial.h:11:29: fatal error: BluetoothSerial.h: No such file or directory
 
and then

compilation terminated.
*** [.pio\build\esp32dev\libabd\OSC\SLIPEncodedBluetoothSerial.cpp.o] Error 1

I'm pretty rookie with code so I don't know what this means.
Any ideas?

Thanks in advance
Dan

alf45tar, 2023-09-06:
The error is related to last version of OSC library. I will release a fix with version 3.1.8

See https://github.com/CNMAT/OSC/issues/139

alf45tar, 2023-09-06:
Version 3.1.8 just released

closed 2023-09-06 by alf45tar

rahji, 2024-01-11:
This error is still occurring. The workaround is to remove the two slipbluetooth header and source files from the PIO libdeps directory. I thought it was because I was using an esp8266 with no Bluetooth but it's happening with esp32 as well.