alf45tar/PedalinoMini #458: 3.1.7 Firmware BUG

bobvc133, 2023-08-25:
1. The length of the interface name is incorrect
2. Tag Settings are not displayed when PC instructions are sent

[image]
[image]

alf45tar, 2023-09-06:
Version 3.1.8 just released fix the 3 bugs added in 3.1.7

bobvc133, 2023-09-07:
> Version 3.1.8 just released fix the 3 bugs added in 3.1.7

Another problem was found that using the sequence function to send Bank Select+/- and Program Change+/- commands is incorrect

alf45tar, 2023-09-07:
Thanks for debugging. I will have a look later today.

alf45tar, 2023-09-07:
> > Version 3.1.8 just released fix the 3 bugs added in 3.1.7
> 
> Another problem was found that using the sequence function to send Bank Select+/- and Program Change+/- commands is incorrect

This is not really bug. It is not yet implemented.

closed 2023-09-08 by alf45tar

bobvc133, 2023-09-09:
> Thanks for debugging. I will have a look later today.

Another problem was discovered. There is no LED color item in the sequence after the configuration is saved.
[image]

alf45tar, 2023-09-10:
Thanks for debugging. Fixed in 3.1.9.