alf45tar/PedalinoMini #461: No smartconfig / WPS option - how to connect?

Jelle7and9, 2023-09-14:
I've upload the firmware to my BPI leaf S3 board. So far so good.
Only I can't use the smartconfig app because it is not supported by newer android versions. 
WPS is also not an option in my network. The board is also not sending a SIDD where I can connect to.

Any solutions for this?
Can I write the SIDD en PASS in the pedalino.h file?  Older issues has pointed to something like this, but the different files are updated and changed over time.

Or is there a newer option for smartconfig? Are new options to connect planned?

alf45tar, 2023-09-14:
You have a lot of alternatives.

1. Boot in AP mode, configure the wireless and reboot in normal mode
2. Use the provisioning via USB
3. Use the provisioning via Bluetooth

Jelle7and9, 2023-09-15:
I can't get into AP mode or provisioning.
Looking at the serial monitor I see this messages:
 
ESP-ROM:esp32s3-20210327
SmartConfig timeout
WPS started
WPS timeout
SYSTEM_EVENT_STA_STOP
SYSTEM_EVENT_STA_WPS_ER_SUCCESS
ESP_ERROR_CHECK failed: esp_err_t 0x3005 (ESP_ERR_WIFI_MODE) at 0x40381ad8
file: "src/WifiConnect.h" line 279
func: void WiFiEvent(arduino_event_id_t)
expression: esp_wifi_coWiFi Provisioning timeout
SmartConfig started
Event: 29
 
I also could't getting into the bootloader anymore, but that I managed with reading the espressif documentation.
"GPIO46 must also be either left unconnected/floating, or driven Low, in order to enter the serial bootloader"
GPOI46 was connected to SDA so that problem is solved.

closed 2023-09-15 by alf45tar