alf45tar/PedalinoMini #464: Creating a new configuration leads to a restart and the configuration is incomplete.

bobvc133, 2023-09-18:
Firmware 3.1.9
Because creating a new configuration will cause a restart, I used JSON Editor to create a complete configuration, which can be imported, applied and saved. However, it will still restart when I create a new configuration again, and the new configuration is still incomplete.

[FM3(complete).txt](https://github.com/alf45tar/PedalinoMini/files/12645231/FM3.complete.txt)
[FM3(incomplete).txt](https://github.com/alf45tar/PedalinoMini/files/12645232/FM3.incomplete.txt)

alf45tar, 2023-10-29:
Can you explain point 2?

bobvc133, 2023-10-29:
> global

Configure the analog pedal in BANK1-20, and the corresponding LED will light up by default, just like adding a (.) ending to the action tag, and set the corresponding LED and color at startup.

bobvc133, 2023-10-29:
> Can you explain point 2?

Configure the analog pedal in BANK1-20, and the corresponding LED will light up by default, just like adding a (.) ending to the action tag, and set the corresponding LED and color at startup.

bobvc133, 2023-10-29:
What is missing in the third question is not only the items in the red box, but also actions such as Repeat and Repeat Overwrite.

bobvc133, 2023-10-29:
In addition, how to use the OSC Message item in the event?
[image]

bobvc133, 2023-11-04:
Another problem was discovered. Starting from firmware 3.1.11, the configuration of the combined button controls is invalid, and the configuration of the single button controls in the combination is also invalid.

bobvc133, 2023-11-20:
7. Regarding the third question, there are still inconsistencies in the codes related to actions and sequence events. Some of the names of schema.json, Config.h, and WebConfigAsync.h are different, and some are missing items. The picture below is a list of my statistics.

- None
- Program Change
- Control Change
- Control Change Snap
- Note On
- Note Off
- Bank Select+
- Bank Select-
- Program Change+
- Program Change-
- Pitch Bend
- Channel Pressure
- MIDI Clock Master
- MIDI Clock Slave
- MIDI Clock Off
- MIDI Start
- MIDI Stop
- MIDI Continue
- Sequence
- Step by Step+
- Step by Step-
- Bank+
- Bank-
- MTC Master
- MTC Slave
- MTC Off
- MTC Time Signature
- MTC Start
- MTC Stop
- MTC Continue
- Tap
- BPM+
- BPM-
- OSC Message
- Profile+
- Profile-
- Set Led Color
- Repeat
- Repeat Overwrite
- Device Info
- Power On/Off

alf45tar, 2023-11-25:
7. Fixed inconsistence in version 3.1.14 (not yet released)

bobvc133, 2023-12-09:
> 7. Fixed inconsistence in version 3.1.14 (not yet released)

8.When editing the pedal pressing mode in jsoneditor, an error message will appear when changing the mode 12L to 1, 2, or 12. There is no problem when selecting the L, 1L, or 2L options.
[image]
[image]
[image]

alf45tar, 2023-12-10:
It is not a PedalinoMini issue but a Json Editor feature explained here how to use it https://github.com/josdejong/jsoneditor/issues/1541#issuecomment-1653919721

bobvc133, 2023-12-24:
> 7. Fixed inconsistence in version 3.1.14 (not yet released)

The action code of the 3.2.0 firmware is still confusing in many places, such as "MIDI Off" and "MIDI Clock Off", "Start", "Stop", "Continue" and "MTC Start", "MTC Stop"," MTC Continue".

[image]
[image]
[image]
[image]
[image]
[image]

bobvc133, 2023-12-24:
[image]
[image]

bobvc133, 2023-12-24:
[image]
[image]

bobvc133, 2023-12-24:
Except for the fourth and seventh problems above, the rest still exist in the 3.2.0 firmware.

alf45tar, 2023-12-24:
Renamed "MIDI off" to "MIDI Clock Off" into `schema.json`.
The rest looks good enough to me.

Did you discovered any bug?

bobvc133, 2023-12-24:
> Renamed "MIDI off" to "MIDI Clock Off" into `schema.json`. The rest looks good enough to me.
> 
> Did you discovered any bug?

https://github.com/alf45tar/PedalinoMini/issues/464#issuecomment-1868563838
[image]

bobvc133, 2023-12-24:
> Renamed "MIDI off" to "MIDI Clock Off" into `schema.json`. The rest looks good enough to me.
> 
> Did you discovered any bug?

1. You can select "MTC Start", "MTC Stop", "MTC Continue" and other events on the action page. After creating a new configuration, JSON Editor will prompt an invalid project because there is no corresponding "MTC Start" in the "schema.json" file. , "MTC Stop", "MTC Continue" and other projects. This is the same problem as the lack of "Repeat" and "Repeat Overwrite" projects mentioned before. Although the action page can select events and apply them and save them to the current configuration, after creating a new configuration, JSON Editor An error will be prompted, and when the application is saved to other BANK, an error will appear that the corresponding event is empty.
2. In the code about the sequence part of the WebConfigAsync.h file, you can see that the event options are "MIDI Start", "MIDI Stop", and "MIDI Continue", but in the relevant condition code, you can see "MTC Start". "MTC Stop", "MTC Continue" and other events, I'm not sure if this is a bug.

[image]

alf45tar, 2023-12-24:
Got it. It will fixed in 3.2.1

alf45tar, 2023-12-27:
Problem 1. I cannot duplicate. I can upload FM3 config file and then save a new configuration without reboot on my T-Display board.

bobvc133, 2023-12-27:
> Problem 1. I cannot duplicate. I can upload FM3 config file and then save a new configuration without reboot on my T-Display board.

After many tests, this problem seems to be caused by the high SPIFFS utilization, because sometimes the new configuration is large, or there are multiple configurations in the configuration list. When the SPIFFS utilization reaches 60%~80%, when the new configuration is created, The probability of a reboot or incomplete configuration is high. Although you can delete other configurations or create new configurations by category, PedalinoMini's 3 user profiles and 21 BANKs will be meaningless because you cannot create so many configurations. In addition, when the BANK takes up a lot, it will have a significant impact on the response speed of the controls. . I don't know if this problem can be improved. In addition, the sixth question is whether it can be fixed in the next version of the firmware. This has a great impact on me. Both the two-button controls and the single-button controls currently cannot be used normally.

alf45tar, 2023-12-27:
Did you considered to increase the SPIFFS size?

alf45tar, 2023-12-27:
Can you clarify "Both the two-button controls and the single-button controls currently cannot be used normally."?

bobvc133, 2023-12-28:
> Can you clarify "Both the two-button controls and the single-button controls currently cannot be used normally."?

This problem has appeared since firmware 3.1.11. Take the following figure as an example. When controls 4, 16, and 6 are all configured with actions, the actions of controls 4, 16, and 6 will not take effect. Delete the action of control 6, and the actions of controls 4 and 16 can take effect normally. Delete the actions of controls 4 and 16, and control 6 can be used normally. 

[image]

bobvc133, 2023-12-28:
> Did you considered to increase the SPIFFS size?

Ready to replace the development board with a larger memory

bobvc133, 2023-12-28:
[image]

bobvc133, 2023-12-28:
[image]
[image]

alf45tar, 2024-01-01:
> > Did you considered to increase the SPIFFS size?
> 
> I am going to replace the 16MB development board. How can I allocate the flash partition size more reasonably? Can you create a partition table for me?

Sure. Which is your board? How many space do you want for SPIFFS?

alf45tar, 2024-01-01:
Assuming BPI-Leaf-S3 here the partition table file `partitions_bpi_leaf_se.csv` for a SPIFFS of 2112 Kb.
```
# Name,   Type, SubType,  Offset,   Size,     Flags
nvs0,     data, nvs,      0x9000,   0x5000,
otadata,  data, ota,      0xE000,   0x2000,
app0,     app,  ota_0,    0x10000,  0x2E0000,
app1,     app,  ota_1,    0x2F0000, 0x2E0000,
nvs,      data, nvs,      0x5D0000, 0x10000,
spiffs,   data, spiffs,   0x5E0000, 0x210000,
coredump, data, coredump, 0x7F0000, 0x10000,
```

bobvc133, 2024-01-02:
> > > Did you considered to increase the SPIFFS size?
> > 
> > 
> > I am going to replace the 16MB development board. How can I allocate the flash partition size more reasonably? Can you create a partition table for me?
> 
> Sure. Which is your board? How many space do you want for SPIFFS?

I am currently using TTGO_T_DISPLAY, BPI_LEAF_S3, and LILYGO_T_DISPLAY_S3, three development boards, with 4MB and 8MB specifications. I plan to replace them all with 16mb SPI flash. The largest single configuration file I need to use is 20 BANKs. If I save 3 user profiles , how much space SPIFFS should be allocated when there are 60 banks in total? Another question is, if I use 16MB for development, will there be enough memory to add the page turning keystrokes function? Many times the music scores I use are in image format, and I use the photo album of the iPad to view them, and there are other applications that do not support MIDI control.

alf45tar, 2024-01-02:
It found this sheet that can build the partition table for you (it is not mine)
https://docs.google.com/spreadsheets/d/1M6_qx00ZR82jyWa2slCMcvLtnXR9Q9dnis5mR5YChW8/edit#gid=0

bobvc133, 2024-01-03:
> It found this sheet that can build the partition table for you (it is not mine) https://docs.google.com/spreadsheets/d/1M6_qx00ZR82jyWa2slCMcvLtnXR9Q9dnis5mR5YChW8/edit#gid=0
> 
> Also this app can be useful https://esp32.jgarrettcorbin.com

I re-partitioned the TTGO_T_DISPLAY 4MB development board and the BPI_LEAF_S3 16MB development board. The following is the partition table.
[image]
[image]

closed 2024-02-08 by alf45tar

bobvc133, 2024-02-16:
> Can you clarify "Both the two-button controls and the single-button controls currently cannot be used normally."?

In 3.2.2 firmware, the problem of control conflict still exists. A button cannot appear in two controls, otherwise both controls will become invalid.

[image]
[image]