alf45tar/PedalinoMini #465: TTGO_T_DISPLAY GPIO set twice

Wapata, 2023-09-23:
Hi !
I wanted to free up my0GPIO27- D3 and GPIO12-D4 for setting DIN_MIDI to theses places.
But **I don't know what to put in the "const byte pin" line** to free theses pin... an non-existant GPIO like GPIO40 ?

And then i've seen that GPIO_NUM_15 is used in two places, DIN_MIDI_IN and FASTLEDS_DATE_PIN
As my LED are working... I suspect that it should be an electrical mess somewhere.

Pedalino.h
```
#elif defined TTGO_T_DISPLAY
#undef  PEDALS
#define PEDALS                8
const byte pinD[] = {GPIO_NUM_25, GPIO_NUM_26, GPIO_NUM_27, GPIO_NUM_12, GPIO_NUM_13, GPIO_NUM_17, GPIO_NUM_35, GPIO_NUM_0};
const byte pinA[] = {GPIO_NUM_36, GPIO_NUM_37, GPIO_NUM_38, GPIO_NUM_39, GPIO_NUM_32, GPIO_NUM_33, GPIO_NUM_35, GPIO_NUM_0};
#define FACTORY_DEFAULT_PIN   GPIO_NUM_0    // Button 2
#define USB_MIDI_IN_PIN       GPIO_NUM_21   // SDA
#define USB_MIDI_OUT_PIN      GPIO_NUM_22   // SCL
**#define DIN_MIDI_IN_PIN       GPIO_NUM_15**
#define DIN_MIDI_OUT_PIN      GPIO_NUM_2
#define BATTERY_PIN           GPIO_NUM_34   // Pin connected to VBAT
#define BATTERY_ADC_EN        GPIO_NUM_14   // ADC_EN is the ADC detection enable port
**#define FASTLEDS_DATA_PIN     GPIO_NUM_15**
```

alf45tar, 2023-09-23:
```
#elif defined TTGO_T_DISPLAY
#undef  PEDALS
#define PEDALS                7
const byte pinD[] = {GPIO_NUM_25, GPIO_NUM_26, GPIO_NUM_39, GPIO_NUM_13, GPIO_NUM_17, GPIO_NUM_35, GPIO_NUM_0};
const byte pinA[] = {GPIO_NUM_36, GPIO_NUM_37, GPIO_NUM_38, GPIO_NUM_32, GPIO_NUM_33, GPIO_NUM_35, GPIO_NUM_0};
#define FACTORY_DEFAULT_PIN   GPIO_NUM_0    // Button 2
#define USB_MIDI_IN_PIN       GPIO_NUM_21   // SDA
#define USB_MIDI_OUT_PIN      GPIO_NUM_22   // SCL
#define DIN_MIDI_IN_PIN       GPIO_NUM_12
#define DIN_MIDI_OUT_PIN      GPIO_NUM_27
#define BATTERY_PIN           GPIO_NUM_34   // Pin connected to VBAT
#define BATTERY_ADC_EN        GPIO_NUM_14   // ADC_EN is the ADC detection enable port
#define FASTLEDS_DATA_PIN     GPIO_NUM_15
```

Wapata, 2023-09-23:
Mmmm, some questions :
As I use PIN 38-A3 and 39-A4 for my two ladders I guess I should not move the pin 39 from Analog no ?
Why pedals from 8 to 7 and not 8 to 6 ?

In the end I should just free up One pin only, GPIO2 is free (and may be my second Midi IN) and then I need an other.
Plus, if i'm okay with that, i guess I can use only One Midi OUT for both interface, it should work to merge theses two electrical connections (i will lose a Midi router rule only..).

What do you think ?

ps : my setup
Pedal 1 - A1-D1 is a pot
Pedal 2 - A2-D2 is a pot
Pedal 3 - A3 is a ladder
Pedal 4 - A4 is a ladder
Pedal 5 - A5-D5 is a jack (so can be everything)
Pedal 6 - A6-D6 is a jack (so can be everything)

alf45tar, 2023-09-23:
```
#elif` defined TTGO_T_DISPLAY
#undef  PEDALS
#define PEDALS                8
const byte pinD[] = {GPIO_NUM_25, GPIO_NUM_26, GPIO_NUM_25, GPIO_NUM_26, GPIO_NUM_13, GPIO_NUM_17, GPIO_NUM_35, GPIO_NUM_0};
const byte pinA[] = {GPIO_NUM_36, GPIO_NUM_37, GPIO_NUM_38, GPIO_NUM_39, GPIO_NUM_32, GPIO_NUM_33, GPIO_NUM_35, GPIO_NUM_0};
#define FACTORY_DEFAULT_PIN   GPIO_NUM_0    // Button 2
#define USB_MIDI_IN_PIN       GPIO_NUM_21   // SDA
#define USB_MIDI_OUT_PIN      GPIO_NUM_22   // SCL
#define DIN_MIDI_IN_PIN       GPIO_NUM_12
#define DIN_MIDI_OUT_PIN      GPIO_NUM_27
#define BATTERY_PIN           GPIO_NUM_34   // Pin connected to VBAT
#define BATTERY_ADC_EN        GPIO_NUM_14   // ADC_EN is the ADC detection enable port
#define FASTLEDS_DATA_PIN     `GPIO_NUM_15```

Wapata, 2023-09-23:
Mouahaha I understand all the changes :-D

Thank you very much for your answers :-) 👍 
Have a great day !

closed 2023-09-23 by Wapata