alf45tar/PedalinoMini #467: Battery + LEDS issue

Jelle7and9, 2023-09-30:
I run firmware 3.1.10 an added a battery connected to JST 2pin connector on a BPI Leaf S3.

1. Battery
The WebUI shows alwas that it's running on battery, also when only powerd via USB.
There is a difference in voltage level. WebUI shows 3.7V and my multimeter shows 4.1V. 
This is the case without BATTERY_PIN 14 connected.

If I add the BATTERY_PIN to +BATT then the oled show's charging battery symbol without _and_ with a battery in de holder.
And the WebUI shows 7.3V (but shows plugged correctly) 
Also when BATTERY_PIN connected and running only on battery  a power reset will result in not be able to connect to my home network. When I remove BATTERY_PIN it instantly connects to my home network.
I have already soldered a new connector to the battery holder to exclude bad connections.
Any thoughts?

2. LED
In the WebUI I can't change the number of LEDS. I can only change this in Pedalino.h?
Is that correct?

Additional info:
PlatformIO shows a couple of issues when I run the inspect command. Don't know of these are important.
Running windows 11 + Visual Studio Code + PlatformIO extension
I only changed PIN 46 to 15 and pin 47 to 16 so the onboard I2C connector works with the OLED.
Furthermore 6 buttons to the first 6 pins[D], potentio to pin 17 and pin 48 for the LEDs

[image]
[image]

alf45tar, 2023-09-30:
1. BATTERY_PIN (GPIO14) is already connected to +BATT using a resistor divider. Do not connect yourself.
https://github.com/BPI-STEAM/BPI-Leaf-S3-Doc/blob/main/sch/BPI-Leaf-S3-Chip-V1.1-SCH.pdf
3. LEDS cannot be changed in WebUI because FastLED library only support static led definition.
https://github.com/FastLED/FastLED/issues/282

Jelle7and9, 2023-09-30:
Thanks for your reply. 
Still I am concernd about the 0.4V voltage difference  in WebUi and measurement with multimeter. (Brymen 789) I don't want to overcharge the battery.

alf45tar, 2023-10-01:
The value measured by board is based on analog to digital conversion and a formula. It is not calibrated at all. The correct amount is your multimeter.
4.1V is ok. The voltage should not be greater than 4.2V or less than 3.2V.

Jelle7and9, 2023-10-01:
Can these errors be ignored? 

[image]

alf45tar, 2023-10-01:
I think so. Most of them are coming from libraries.

Jelle7and9, 2023-10-02:
Is there a workaround for calibration and cut off for over- or undercharge protection? Auto power off?
And how to power off the PedalinoMini when battery attached? It never powers off unless I detach the battery. That is not so fine when everything is build in ;-) 
Power Off in WebUI results in reboot. It directly powers on again.

alf45tar, 2023-10-02:
The fix the calibration change the following formula

https://github.com/alf45tar/PedalinoMini/blob/fb77242fba1fd566a05f7aff69078a048bba0faa/src/PedalinoMini.cpp#L199

There are no risk of over/under charging because the battery recharge is managed by [MCP73831](https://ww1.microchip.com/downloads/en/DeviceDoc/MCP73831-Family-Data-Sheet-DS20001984H.pdf) chip.

To power off PedalinoMini use a switch to cut off the battery power or the Power Off action to put the board into deep sleep.

rigr, 2023-10-02:
In another project I finally got to this method to show the voltage:
batteryV = ((float)analogRead(14) / 4095) * 7.26;
It shows 4.2V and 3.6V like my multimeter. I use a small battery from a broken mini cam.

Jelle7and9, 2023-10-03:
I will try to make litle adjustments in the code soon. 
And somehow deep sleep mode don't work. Something triggers an wake up / start up. No idea right know. Floating pins perhaps?

Jelle7and9, 2023-10-04:
> The fix the calibration change the following formula
> 
> https://github.com/alf45tar/PedalinoMini/blob/fb77242fba1fd566a05f7aff69078a048bba0faa/src/PedalinoMini.cpp#L199
> 
> There are no risk of over/under charging because the battery recharge is managed by [MCP73831](https://ww1.microchip.com/downloads/en/DeviceDoc/MCP73831-Family-Data-Sheet-DS20001984H.pdf) chip.
> 
> To power off PedalinoMini use a switch to cut off the battery power or the Power On/Off action to put the board into deep sleep.

I changed the code to:
`uint16_t voltage = ((uint32_t)v * 2.25 * 33 * vref) / 10240;`
Now its more accurate. I don't know this is the right way.

Jelle7and9, 2023-10-04:
Also sometimes I get these errors in the terminal when building. It has something to do with the platformio packages and platforms I think.
I can't get an constant building output. Any thoughts where to look or configurate?

[image]
[image]
[image]

alf45tar, 2023-10-04:
Yellow message are just warnings.
Errors are in red.

Jelle7and9, 2023-10-08:
Allright, with some trial and error.

When I change the value 10240 to 8850 the battery voltage is showed like the metering does and battery indicator on oled look good.
`uint16_t voltage = ((uint32_t)v * 2 * 33 * vref) / 8850;`
Changing the vref will give strange outcome.. Althoug Vref is between 1000 and 1200 margin conform ESP32 documentation.

The dividing value does the calibration likely.
I have read data with serial.print and it seems a voltage of 4,17V is equal to battery_pin value of ~2070 (out of 4095)

closed 2023-10-23 by Jelle7and9