alf45tar/PedalinoMini #468: New design

alf45tar, 2023-10-04:
@joegriffinable please add here your requisition for the customized PedalinoMini Case1 design

joegriffinable, 2023-10-04:
Hello,
Not much change from the original design,
I have these to fit inside:
[image0.jpeg]

I tried to print the original version and came across a few issues:
Couldn’t fit the screen in, couldn’t fit MIDI ports.
I’ve bought an arduino mega to fit in and also a Battery case/charge.

The original design was slightly too big,
Please could the dimensions be 8cm/20cm (ok with the height at 3/4cm)

Joe Griffin

jisv48, 2023-10-04:
Here is my requisition for the original case...Hopefully indicates that, just lt me know how much it will cost.  Cheers

alf45tar, 2023-10-04:
PedalinoMini Case 1 is designed for TTGO TDisplay only which include the battery recharge too.
Arduino Mega is not supported by PedalinoMini.

Dimension cannot be reduced to 8 cm (check the images). There is no space to allocate for the display.

joegriffinable, 2023-10-04:
Thanks,
I’ll scrap the mega arduino, I thought I could get midi over usb through it.
Is there a way of extending the width of the pedal to 25cm(max.) and reducing to 8cm to accommodate the screen and buttons? Im just trying to squeeze it into a very compact pedalboard!
Thanks

Joe Griffin

alf45tar, 2023-10-06:
The minimum dimension are 265 x 80 mm.
Where can I send/upload the files?

joegriffinable, 2023-10-08:
Thanks,
I’m not sure where to send them! How do you normally do it?
Thanks

Joe Griffin

The minimum dimensions are 265 x 80 mm.

alf45tar, 2023-10-08:
I can send by email or if you want to share your design with sponsors I can upload on github.

joegriffinable, 2023-10-08:
I’m happy with sharing!
Please upload on GitHub,
Cheers

Joe Griffin

alf45tar, 2023-10-08:
https://github.com/pedalino-sponsors/PedalinoMini/tree/master/case1

closed 2023-10-26 by alf45tar