alf45tar/PedalinoMini #469: Issues 3.1.10 firmware

Jelle7and9, 2023-10-20:
Hi Alf45ar,

Have some strange behavior with my latest build. ( erased flash and clean build)
1. Actions are sometimes duplicated and everything is getting messed up. (windows 11 + chrome browser)
2. WebUI on mobile the action tab is messed up. Action numbers are moving under my fingers. I can't select one.
3. I have a ladder 6 button configuration. Only buttontag action 1 is shown on display. Nothing from the other 5 buttontags.
4. The bank name is on the same spot as the battery indicator when no colon before bank name. (when USB powered there is blinking battery indicator)
5. When battery and USB powered no battery indicator at all.
6. Only on battery power no battery indicator, battery measures 3.55V
   Can ADC_2  (GPIO_14) + Wifi be the problem for the strange battery level and indicator behavior?
8. I find it hard to see what configuration is loaded to profile A, B or C
9. Reboots when applying or apply and safe an other configuration.
10. One time direct wifi connection, next time timeout errors.

My config is very basic:
Pedal 1 - 6 button ladder - 10K resistor on pin A1 - values 0 - 158 - 321 - 492 - 655 - 831 - 1023
If I move Pedal 1 to Pedal 4 and change to pin A4 then the tag is under the bank name. Also only the first button
Pedal 2 - momentary - button form rotary encoder pin D2
Pedal 3 - Jog wheel - rotary encoder - 10K resistor on pin A3
Pedal 9 - boot button - pin 0
Firmware 3.1.10 no adjustments in code. Straight cloned from Github.

[image]

[image]

[image]

https://github.com/alf45tar/PedalinoMini/assets/137293236/c1b8ef5f-b4a5-4b41-bf64-d34c819ec907

[image]

alf45tar, 2023-10-21:
Which board are you using?

Jelle7and9, 2023-10-21:
BPI leaf S3.

Jelle7and9, 2023-10-21:
Maybe my board is not working properly anymore.
I ordered a second one. So I can exclude a bad power supply input and/or a bad functioning ADC.
Can this be the cause for the wifi en battery indicator problems?

Also the Espressif documentation says this;
Since the ADC2 module is also used by the Wi-Fi, reading operation of [adc2_get_raw()] may fail between [esp_wifi_start()] and [esp_wifi_stop()]. Use the return code to see whether the reading is successful.

Wifi timeouts could also be related to my network or nearby wifi networks. Today wifi works like a charm.

alf45tar, 2023-10-21:
Why do you not consider the LilyGo T-Display S3?

alf45tar, 2023-10-21:
> Maybe my board is not working properly anymore. I ordered a second one. So I can exclude a bad power supply input and/or a bad functioning ADC. Can this be the cause for the wifi en battery [...]

It usually a software issue.

Jelle7and9, 2023-10-21:
Programming controls gives strange behavior
For example, when I fill in control 1 to 6 in bank 3. When I apply and save only control 1 is saved and control 2 to 6 are empty again. So I have to program 1 control at the time and save before programming the next control.

alf45tar, 2023-10-21:
Can you attach the log?

Jelle7and9, 2023-10-22:
This log is when applying a config to a profile and reboot happen.
[esp-web-tools-logs (23).txt](https://github.com/alf45tar/PedalinoMini/files/13063021/esp-web-tools-logs.23.txt)

Looks like there are sometimes false readings from pedal1 button 5 and 6. Although this was not the case yesterday.
I measured also 2.9 ~3.0 V on ladder buttons. The breadboard has resistance. Today connected  directly to 3.3 pin and is gives 3.2V.

[esp-web-tools-logs (26).txt](https://github.com/alf45tar/PedalinoMini/files/13063042/esp-web-tools-logs.26.txt)
[esp-web-tools-logs (25).txt](https://github.com/alf45tar/PedalinoMini/files/13063043/esp-web-tools-logs.25.txt)
[esp-web-tools-logs (24).txt](https://github.com/alf45tar/PedalinoMini/files/13063044/esp-web-tools-logs.24.txt)

[image]

[image]

Jelle7and9, 2023-10-23:
I completely rebuild my setup on a breadboard.
I use other pins now and edited the platformio.ini, now turned off -D BATTERY
SMART CONFIG and WPS I always disable.
I Skipped the pins for pedal 1 to 4 and use 5 to 9.
10K Pull up resistors for pin A8 (ladder) and pin A5 Jog Wheel.

Also one suggestion, do not use GPIO_46 as this has to be low to enter boot mode.
The public version is fine, the sponsor version has this GPIO pin as SDA pin configured.

When I read the logs there are some errors.
1. [ 20193][W][WiFiGeneric.cpp:1057] _eventCallback(): Reason: 8 - ASSOC_LEAVE
Terminal disconnected: NetworkError: The device has been lost.
2. [250018][W][AsyncTCP.cpp:950] _poll(): rx timeout 4
[260716][W][AsyncTCP.cpp:930] _poll(): pcb is NULL
3. Memory fragmentation warning: webpage memory allocation 8302 bytes greater then 8192 bytes reserved.

With this build and own created configuration I did not get actiontags at all displayed on the OLED.
If I change to Blueboard config it shows nicely all the tags.

Logs from this morning:
[esp-web-tools-logs (31).txt](https://github.com/alf45tar/PedalinoMini/files/13067815/esp-web-tools-logs.31.txt)
[esp-web-tools-logs (30).txt](https://github.com/alf45tar/PedalinoMini/files/13067816/esp-web-tools-logs.30.txt)

alf45tar, 2023-10-23:
Thanks for the logs.

1. Need investigation
2. More details here but it is not an issue. https://github.com/me-no-dev/ESPAsyncWebServer/issues/534
3. I will fix it

alf45tar, 2023-10-24:
```
1. Actions are sometimes duplicated and everything is getting messed up. (windows 11 + chrome browser)
```
```
2. WebUI on mobile the action tab is messed up. Action numbers are moving under my fingers. I can't select one.
```
Try version 3.1.11
```
3. I have a ladder 6 button configuration. Only buttontag action 1 is shown on display. Nothing from the other 5 buttontags.
```
```
4. The bank name is on the same spot as the battery indicator when no colon before bank name. (when USB powered there is blinking battery indicator)
```
Try version 3.1.11
```
5. When battery and USB powered no battery indicator at all.
```
```
6. Only on battery power no battery indicator, battery measures 3.55V
Can ADC_2 (GPIO_14) + Wifi be the problem for the strange battery level and indicator behavior?
```
```
7. I find it hard to see what configuration is loaded to profile A, B or C
```
Not clear the request. Please add more details.
```
8. Reboots when applying or apply and safe an other configuration.
```
Try version 3.1.11
```
9. One time direct wifi connection, next time timeout errors.
```

Jelle7and9, 2023-10-26:
Thanks Alf45tar!

I just build this firmware 3.1.11
1. Pedals, Controls and Actions are programmable, nothing strange. (Windows + Chrome)
2. WebUI on my phone, under tab ACTIONS: I can't select banks. Bank numbers are movable under my finger. Control All and Actions are perfectly selecting and editable. Samsung S10+ and Chrome, Android 12, Kernelversion 4.14.113-25257816
See screenshot.
3. Ladder config is not showing any tags at all. Also Tags from Momentary and so on are gone.
   (.) before Bank Name is showing nothing. When I press a button I see the tag and value on OLED.
4. Looks solved, I see only the battery indicator, no Bank Name in upper right corner anymore.
5. / 6. / 7.  Witch point is not clear for you? No battery connected right now. I will try this later on.
8. Still sometimes 

reboots when Apply & Save configuration.
9. Wifi looks like more stable now. I restart all my AP's and set UAPSD and BSS transition off. (UniFy Network)

Still some errors with memory fragmentation.
Can memory fragmentation makes it unstable? I did turned of all interfaces, except USB midi out.
See recent logs.

I have an extra brand new BPI leaf S3 now. I will try this new one in the next few days to exclude errors in the hardware.

[esp-web-tools-logs (35).txt](https://github.com/alf45tar/PedalinoMini/files/13182573/esp-web-tools-logs.35.txt)
[esp-web-tools-logs (34).txt](https://github.com/alf45tar/PedalinoMini/files/13182574/esp-web-tools-logs.34.txt)
[esp-web-tools-logs (33).txt](https://github.com/alf45tar/PedalinoMini/files/13182575/esp-web-tools-logs.33.txt)
[esp-web-tools-logs (32).txt](https://github.com/alf45tar/PedalinoMini/files/13182576/esp-web-tools-logs.32.txt)
[Upload and Upload Filesystem Image  .txt](https://github.com/alf45tar/PedalinoMini/files/13182816/Upload.and.Upload.Filesystem.Image.txt)

[image]

Jelle7and9, 2023-10-27:
With the **brand new** BPI leaf S3 I have so far the same 2 issues with 3.1.11 firmware.
Noting connected to any GPIO_pin. Build without Smartconfig and WPS

WebUI on mobile device, Action tab, Bank selection is nog working. I booted as AP config and connect to PedalinoMini wifi and
Memory fragmentation warning: webpage memory allocation 8610 bytes greater then 8192 bytes reserved, and
Memory allocation differs from my other build. It is around 8700 bytes, more or less.

Just wanted to check profile A B C and connection lost. Also eventtype:press; pedalbutton 4/2 can't be happend. I have nothing connected at all.

Looking for configuration files on SPIFFS root ...
/blueboard.cfg
/case1.cfg
/mustang.cfg
/Jelle_build 27-10-2023.cfg
done.
Looking for configuration files on SPIFFS root ...
/blueboard.cfg
/case1.cfg
/mustang.cfg
/Jelle_build 24-10-2023.cfg
done.
Looking for configuration files on SPIFFS root ...
/blueboard.cfg
/case1.cfg
/mustang.cfg
/Jelle_build 24-10-2023.cfg
done.
[645237] Pedal/Button: 4/2     EventType: Press             ButtonState: 0
[670092][W][AsyncTCP.cpp:950] _poll(): rx timeout 4
[944594][W][AsyncTCP.cpp:950] _poll(): rx timeout 4
[944596][W][AsyncTCP.cpp:950] _poll(): rx timeout 4
[1102515][W][AsyncTCP.cpp:950] _poll(): rx timeout 4
[1102518][W][AsyncTCP.cpp:950] _poll(): rx timeout 4
[1574260][W][AsyncTCP.cpp:950] _poll(): rx timeout 4
[1574264][W][AsyncTCP.cpp:950] _poll(): rx timeout 4
[1581263][W][AsyncTCP.cpp:950] _poll(): rx timeout 4
Looking for configuration files on SPIFFS root ...
/blueboard.cfg
/case1.cfg
/mustang.cfg
/Jelle_build 27-10-2023.cfg
done.
Looking for configuration files on SPIFFS root ...
/blueboard.cfg
/case1.cfg
/mustang.cfg
/Jelle_build 24-10-2023.cfg
done.
[1704945][W][AsyncTCP.cpp:950] _poll(): rx timeout 4
Terminal disconnected: NetworkError: The device has been lost.

alf45tar, 2023-10-28:
WebUI on my phones (iOS and Android) works as expected. Banks can be moved to sort them. It is a feature not a bug. 

Memory fragmentation on short term is not an issue. It can reduce the stability on long term when large block of memory cannot be allocated.

bobvc133, 2023-10-28:
> WebUI on my phones (iOS and Android) works as expected. Banks can be moved to sort them. It is a feature not a bug.
> 
> Memory fragmentation on short term is not an issue. It can reduce the stability on long term when large block of memory cannot be allocated.

Can you take a look at my question? #464

alf45tar, 2023-10-28:
> > WebUI on my phones (iOS and Android) works as expected. Banks can be moved to sort them. It is a feature not a bug.
> > Memory fragmentation on short term is not an issue. It can reduce the stability on long term when large block of memory cannot be allocated.
> 
> Can you take a look at my question? #464

I am looking into it.

alf45tar, 2023-10-29:
GPIO14 is connected to +VBATT on BPI Leaf S3. There is no way to discover when connected to USB power without using additional hardware and another GPIO.

Wapata, 2023-11-03:
> BPI leaf S3.

Hi,
Excuse me but how di you compil the code ? 
I've got this message whatever the version of the firmware and don't know how to resolve it...
```
<command-line>: fatal error: /.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/arduino_tinyusb/include/tusb_config.h: No such file or directory
compilation terminated.
*** [.pio\build\bpi-leaf-s3\src\PedalinoMini.cpp.o] Error 1
```

alf45tar, 2023-11-04:
Are you using Windows? If yes set the HOME env varaiable

`set HOME=%HOMEDRIVE%%HOMEPATH%`

Wapata, 2023-11-04:
I feel bad... I don't have a clue of what you mean, and google is very bad at helping me with visual studio... like it's the first project ever where i so dont understand the answers that i doubt of my questions...

So, i've made ctrl+shift+P and then past you line, then i have the "welcome to plateformIO" page and don't know what to do next.

I'm sorry to bother you :'-(

Jelle7and9, 2023-11-04:
Look in platformio.ini for your board matching build_flags.
Change the line with the USB CONFIG_FILE
-D  CFG_TUSB_CONFIG_FILE="/**_users/your_user_name_**/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/arduino_tinyusb/include/tusb_config.h"'

Wapata, 2023-11-04:
Thank you for the clarification. One step is made...
The next is 
`.pio/libdeps/bpi-leaf-s3/ESP32TinyUSB/src/sdusb.h:4:10: fatal error: SD.h: No such file or directory`

I wonder why the compilation worked 4 month ago and not now o_O it's slowly driving me crazy

Jelle7and9, 2023-11-05:
Did you saved and _staged_ the changes in the ini file?

Wapata, 2023-11-05:
I've deleted everything and "get" the project from Visual Studio and not by downloading it. Made the "user" line change and then it worked.
Thank you for your kind help !

Jelle7and9, 2023-11-05:
If you get this error message, you can ignore it.
Or edit the OSC library.json and delete the left bracket character in line 24. 

[image]

Jelle7and9, 2023-11-27:
What is the status of the ladder config + buttontags issue?

Jelle7and9, 2023-12-17:
Somehow with 3.1.13 firmware the ESP Webtool wil not connect when pressing "visit device", only to direct IP adress it wil be accessible. 
Also the OLED is working fine when upload the firmware, but as soon I made changes in the the config, after a reboot he OLED is off and not showing anything.

Any thoughts?

Jelle7and9, 2024-01-23:
Webtool and Visit issue has been related to multicast/broadcast settings in my network.  Thats solved.
Running 3.2.0 now and looks to work better for me.

closed 2024-01-23 by Jelle7and9