alf45tar/PedalinoMini #470: Switches not working

mrybas, 2023-10-26:
Hi! I assembled everything according to the scheme https://github.com/alf45tar/PedalinoMini-Case-1 and the bottom row of buttons (1a,2a,3a,4a) work perfectly. But the top buttons don't work.
 Assembled on the board TTGO T-DISPLAY

alf45tar, 2023-10-26:
did you load the case1 configuration file?

alf45tar, 2023-10-26:
Pedals from 1 to 4 need to be configured as Latch 2.

mrybas, 2023-10-26:
yes, I'm  loaded the case1 configuration file. I'm configured pedals from 1 to 4 to be configured as Latch 2, now pedals 1-4, and 10 working. But 7-9 still not work 
Here is my setting
[image]
[image]
[image]
[image]

mrybas, 2023-10-26:
I completely rebuilt the entire circuit and now all the buttons work perfectly. It looks like the problem was a bad connection somewhere on the +3 volt line. Everything works great! Thanks for the help!

closed 2023-10-26 by mrybas