alf45tar/PedalinoMini #471: PedalinoMini Case1 new design

alf45tar, 2023-10-27:
@iammac360  please add here your requisition for the customized PedalinoMini Case1 design

iammac360, 2023-11-06:
sorry @alf45tar I was busy this pass few days, I didn't notice your invite right away on the pedalino-sponsors repository and the link expired, can you reinvite me again? 

for the design requisition, I'm good with the default dimensions but I just want to know if it is possible, you make the screen on the center with the screen size of 1.3 inch? planning to use doit esp32 devkit v1 and a 1.3inch ssh1106 display.

alf45tar, 2023-11-18:
I was abroad during the last 2 weeks and I was disconnected. 
Re-sent the invitation.
PedalinoMini Case1 is for T-Display only.

zavorra, 2023-12-12:
Hi, sorry to hijack this thread, it seems that it is no longer possible to send direct messages on github... 

I will receive a T-Display soon to build a pedalino, so I just became a sponsor. Could you please add me to the restricted repo? 

Thanks in advance

closed 2024-02-08 by alf45tar