alf45tar/PedalinoMini #472: Schematic Schematic_PedalinoMini_Sheet-1.svgnot up to date

Wapata, 2023-11-05:
Hi there !
This schematic seem not up to date : Schematic_PedalinoMini_Sheet-1.svg at least for Leaf S3 board. As SDA and SCL as moved to there SDA-SCL "natural" pin, Pin_A5 and A6 can't be here too.
There is other few things not up to date into the Readme page, where is it easier for us to tell you what ?

Have a nice day !

Wapata, 2023-11-05:
[image]

And here GPIO_15 and 16 are assigned to A5/A6 and also SDA/SCL

Jelle7and9, 2023-11-05:
I am also a little bit confused, cause I do not know what the reason was to change PIN assignments. Although I have very much respect for Alf45tar and all the effort he puts in this. And with different type of ESP32 boards it is sometimes challenging.

If you want to use the onboard 4pin I2C socket then pin 15 and 16 are the default SDA and SCL pins. So that is logical.
For the previous firmware config I only changed SDA-46 to pin 42. (46 is strapping pin and has te be low for download boot mode)
JTAG pins (39-42 )are not used by PedalinoMini and these pins are free assignable GPIO pins I think.

What I wonder if for example GPIO_15 an 16 can be used as output and input simultaneously. Output for SDA/SCL (read) and input signals for pedal. Quick search on Google learns it is possible.
If so then the latest pin assignment have logic, it counts up for pinD[ ] and down for pinA[ ]
I can easily change the connections in my test setup on a breadboard to test.

Wapata, 2023-11-05:
[image]

Using "Momentary 1" kind look like a shortcut to me, from Pin_A1 linked to +3.3v via 10k resistor to Pin_A1 linked to GND.
I'm not able to have any reaction from my buttons by using "Momentary 1" but they work great with "Momentary 2"

alf45tar, 2023-11-23:
Digital pins are using the internal pull-up resistor activated by the following code:

`pinMode(PIN_D(i), INPUT_PULLUP);`

There is no shortcut. When the switch is OPEN the pin is HIGH and vice versa.

alf45tar, 2024-02-08:
Update schematics

closed 2024-02-08 by alf45tar