alf45tar/PedalinoMini #473: Cannot save or load configurations on S3

maarthome, 2023-11-09:
Hi Alf45tar,

I noticed that I cannot save or load configurations on the S3...? It would be great if this is made possible again.
Also only 1 LED is defined for the BPI_LEAF_S3 in 'Pedalino.h':

#elif defined ARDUINO_BPI_LEAF_S3           // https://wiki.banana-pi.org/BPI-Leaf-S3
#undef  LEDS
#define LEDS                  1

Thanks for your incredible work!

Regards,
Maarten

Designbase, 2023-11-09:
Dank je voor je bericht!

Ik ben met vakantie tot en met 20 November.

De mail wordt gelezen en mochten er calamiteiten zijn dan zal ik actie ondernemen!

Met vriendelijke Groet,

Ronald Westland
Designbase

Jelle7and9, 2023-11-10:
Same for me also. Firmware 3.1.13 and BPI leaf S3

maarthome, 2023-11-21:
Hi Alf45tar,

I recompiled the code and now it is OK :-)
I can load and save configurations!!

Thanks again for this great project!

Regards,
Maarten

closed 2023-11-22 by maarthome