alf45tar/PedalinoMini #474: Using SH1106 OLED 1.3 128x64 Display with web installer ESP32

WernAir77, 2023-11-11:
Hi,

I tried to use a SH1106 OLED display with the web installer on a ESP32, like for ESP32_DOIT_DEV_Kit_1, but I only see a small clear line downside and the rest is snowy ;) 

I tried with and without pull ups and 5V instead of 3v3, but nothing changed. 

Is there a solution? For some years I installed pedalino with VS Code and could change between SD1306 and SH1106, but actually I can't use this workaround, because many libs are missing when I clone from git and follow the description.

WernAir77, 2023-11-12:
Today I tried the TTGO T-EIGHT 3.1.6 and what should I say, the display works fine but I can't get any access to the web interface or wifi config. 
I also tried several versions of the ESP_32_DOIT_DEV_Kit_1 but with the same result, the display shows a small line and the rest is snowy.

Jelle7and9, 2023-11-12:
One condition is that you build, upload and _upload filesystem image_ in PlatformIO. The last one is important for the correct WebUI experience.
Do you connect to the WebUI via the Pedalino installer https://alf45tar.github.io/PedalinoMini/installer
It is explained in the Readme

WernAir77, 2023-11-12:
Yes, I connect via the WebUI after installation of DEV_DOIT_Kit_1, with TTGO_T-EIGHT, I was not able to connect Wifi.

WernAir77, 2023-11-14:
Sorry for spamming, but I'm lost.
Were can I find bluetoothSerial.h, FastLED.h, Improv.h, HSCR04.h, and so on? I just follow the instruction but these libs are missing and I don't know were to search for...

wcraffonara, 2023-11-14:
Hi,
  I also tried to compile the project and I've exactly the same issues with missing libs. It would be very nice to have some instruction how to get the libs. Thanks!
Regards

alf45tar, 2023-11-16:
The lib issue has ben fixed in sponsor version since long time. It will be fixed in public version too when one of the goal will be reached.

closed 2023-11-17 by WernAir77